FS-Driverless/Formula-Student-Driverless-Simulator #1: Setup the Unreal Engine project

SijmenHuizenga, 2020-05-02:
Add the UE4 project to this repository. We have to research best practices regarding UE4 projects in git/github. 

- [ ] Create a new project, give it a nice name and add it to git
- [ ] Populate it with an (empty) airsim project
- [ ] Check the license for all added content 
- [ ] Add any required attributions in the readme
- [ ] Document how to get this project running on a new machine, including airsim documentation references.

davidoort, 2020-05-07:
Seems like all the checkboxes are done. I will test that I can launch the Unreal Editor with the technion car from here

davidoort, 2020-05-07:
okay, I'll give you some time before testing this

SijmenHuizenga, 2020-05-08:
I believe you tested it right @davidoort?

davidoort, 2020-05-08:
YES! I am now testing the project from this repo with the Unreal Installation on the old GCP instances. This task is done

closed 2020-05-08 by davidoort