FS-Driverless/Formula-Student-Driverless-Simulator #2: Car Dynamic Model

SijmenHuizenga, 2020-05-02:
Airsim uses [this code](https://github.com/microsoft/AirSim/tree/master/Unreal/Plugins/AirSim/Source/Vehicles/Car) to move the car according to inputs. It is pretty good but we require an individual four-wheel motor model. The input must be 4x motor torques and 1x steering angle. We have to create our own by copying this model and changing it.

SijmenHuizenga, 2020-05-08:
We concluded that creating a 4-wheel torque dynamic model is quite a lot of work compared to the gains. We also suspect that some teams run their algorithms that calculate individual wheel torques based on targeted speed and direction separate from their autonomous systems, maybe even on embedded system. It therefore could be a lot of work for teams to move this code to be part of their simulated autonomous systems.

Because of all this uncertainty we have decided to hold this task for now.

davidoort, 2020-05-09:
One of the things I would add to this investigation is to switch the current vehicle model from combustion (geared) to electric.

davidoort, 2020-05-28:
The Unreal Engine repository contains (as a third party library) the code for [PhysX](https://docs.nvidia.com/gameworks/content/gameworkslibrary/physx/guide/Manual/Vehicles.html) vehicles that was developed by Nvidia. Airsim simply interacts with the PhysXCar API in [these](https://github.com/FS-Online/Driverless-Competition-Simulator/tree/93fa784106c70e53e973e6d755ab5430feeef9da/UE4Project/Plugins/AirSim/Source/Vehicles/Car) files, which have configurable parameters exposed in the uassets files [here](https://github.com/FS-Online/Driverless-Competition-Simulator/tree/master/UE4Project/Plugins/AirSim/Content/VehicleAdv/SUV)

closed 2020-06-04 by SijmenHuizenga