FS-Driverless/Formula-Student-Driverless-Simulator #5: Lap timer and counter

SijmenHuizenga, 2020-05-02:
Implement a lap timer that tracks how long the car did for every lap. 

The formula student 2020 rulebook writes:

> D 8.2.5 Starting - A go-signal from RES is used to indicate the approval to begin. Timing starts after the vehicle crosses the starting line.

Let's use the center point of the car as the measurement point for timing. So start and stop the timer when the car's center point crosses the start/finish line.

A time value for every lap must be recorded and shown on the heads up display.

A lap counter can be derived from the number of timed laps. This value must also be shown on the heads up display.

Transmitting the lap times and lapcounter to some place outside unreal engine is out of scope for this issue.

Riliano, 2020-05-14:
A working prototype I have times the lap by setting up a line between the big orange cones. Once the car has completely passed the line it counts as a lap and the time is stored. I believe it makes sense to include exporting this data inside the scope of the issue, as that influences how the data is stored in the first place.

I'm thinking of making a static array or an std::vector inside RefereeState where the time of each lap will be stored, as well as another field which will store the time of the entire run.

SijmenHuizenga, 2020-05-14:
Zsofia from Fs-Online and I discussed how the timer and finishline should interact. The rules will be as follows: The lap counter starts and stops when the nose of the car crosses the start/stop line.

We think the best livestream experience.

PR #36 (2020-05-17): Lap counter
Adds an object to represent start/finish line and exports the time of each passed lap.
Closes #5
closed 2020-05-17 by Riliano