FS-Driverless/Formula-Student-Driverless-Simulator #13: F10 spectator view streaming

SijmenHuizenga, 2020-05-03:
Is it possible to stream camera footage from inside unreal engine to some external streaming platform?

Add a new camera inside the unreal engine world that is not coupled to the car. This camera can just be statically located with a top-down view of the track. What this camera sees must be streamed to some third party. Go online, find some online streaming platform (youtube? twitch?) and get this camera view live in their. 

The goal of this task is to figure out how streaming to some external location works. We will need this for the livestream and getting an understanding of how this works is essential.

SijmenHuizenga, 2020-05-10:
I did some research and I couldn't find any resources on how to do a livestream. I have come to the conclusion that we shouldn't try to livestream within ue, instead just capture the display and use specialized software to make the video available to the rest of the world. We can just configure the primary camera view to be super nice and can be directly controlled by some user. 

I looked at: 

* [Enable Live Stream](https://docs.unrealengine.com/en-US/BlueprintAPI/Game/EnableLiveStreaming/index.html) shows something DVR related. But other than that page there is no docs at all.
* [TwitchWorks](https://www.unrealengine.com/marketplace/en-US/product/twitchworks) only does meta-things like chat and interactions, it does not do actual video streaming
* [NDisplay](https://docs.unrealengine.com/en-US/Engine/Rendering/nDisplay/index.html) feels overkill and does not solve the problem of sending images externally

sanzgadea, 2020-05-10:
There's this plugin for Unreal Engine called [Pixel Streaming](https://docs.unrealengine.com/en-US/Platforms/PixelStreaming/index.html) but it only seems to work for computers in the same network. For example you can stream from UE and view it in a local browser window. We need to find out of to send that video feed to an online streaming service.

SijmenHuizenga, 2020-06-16:
The current plan is to setup a simple multiplayer system where the multiplayer client is a view-only camera controlled by the keyboard. This way the compute-expensive rendering happens on the spectator's computer and the simulator is not impacted.

closed 2020-06-21 by SijmenHuizenga