FS-Driverless/Formula-Student-Driverless-Simulator #7: Reset world

SijmenHuizenga, 2020-05-03:
When the airsim reset rpc call is triggered the following must happen:

- The car must be placed on the start location, velocity, torque and steering must be set to 0.
- All cones must be placed upright to their original location
- The referee status must be reset. Number of DOO cones must be set to 0, laptimes list must be cleared and the driving state must be set to NOT_STARTED

After this reset has finished the whole system must be ready to do another run.

davidoort, 2020-05-08:
You can use the ROS wrapper for most of these things by running the service call `rosservice call /airsim_node/reset "waitOnLastTask: false" `

SijmenHuizenga, 2020-05-14:
Good find! Reset is indeed the signal that we can use. We do however need to reset cones and referee state.

SijmenHuizenga, 2020-05-28:
We will not be resetting the world. Instead we will restart unreal engine between runs.

closed 2020-05-28 by SijmenHuizenga