FS-Driverless/Formula-Student-Driverless-Simulator #8: Dynamically update car sensor suite

SijmenHuizenga, 2020-05-03:
The airsim rpc api must be extended with a new function that sets the sensor suite. For this first version this includes:

- Camera mounting on the car (x, y, z, roll, yaw, pitch) 
- Lidar mounting on the car (x,y,z, roll, pitch, yaw)
- GPS position on the car (x, y, z)

davidoort, 2020-05-08:
This script might be useful for doing this https://github.com/microsoft/AirSim/blob/master/ros/src/airsim_tutorial_pkgs/scripts/multi_drone_json_creator.py

skimslozo, 2020-05-17:
Current PR: https://github.com/FS-Online/Driverless-Competition-Simulator/pull/50
Next task: checking feasibility of the whole settings.json in-sim update, go/no-go on dynamic updates by the end of this week

closed 2020-05-26 by SijmenHuizenga