FS-Driverless/Formula-Student-Driverless-Simulator #9: Simulator v1

SijmenHuizenga, 2020-05-03:
The simulator program will be the 'head quarters' of this system. The competition officials will be able control the events using a gui provided by the simulator, the autonomous systems will connect to the simulator to retrieve sensor data and send trajectory setpoints. 

The challenge in this task is to create the first version of this program. 

It must be able to connects to airsim and retreives lidar + camera stream from the car. It must wait for a start signal from a webinterface. When the start signal is received the camera and lidar stream must be forwarded to an external autonomous system through ros. Publishing on a ros topic is good enough for now. The simulator must listen on a ros topic for car setpoints and forward those to the car inside airsim.

This program must be built in python.

davidoort, 2020-05-04:
You mentioned that a way to achieve this would be to build on/modify the ros wrapper for airsim, however that is written fully in C++, so I guess using C++ is not out of the question?

SijmenHuizenga, 2020-05-05:
> You mentioned that a way to achieve this would be to build on/modify the ros wrapper for airsim, however that is written fully in C++, so I guess using C++ is not out of the question?

Sorry! I must have misremembered this. If you guys want to do C++ i am fine with that but if you want to rewrite the useful parts to python that's also fine with me.

SijmenHuizenga, 2020-05-10:
This shouln't be closed

davidoort, 2020-05-11:
Waiting for @mvanlobensels to close this. The ROS part is done

SijmenHuizenga, 2020-05-12:
First version was merged in #20, follow up tasks are #28 and #29

closed 2020-05-12 by SijmenHuizenga