FS-Driverless/Formula-Student-Driverless-Simulator #14: VNC dying when opening VS Code

davidoort, 2020-05-06:
[image]

mvanlobensels, 2020-05-08:
Same problem. This is highly annoying indeed

SijmenHuizenga, 2020-05-14:
Trying to find the root cause of this problem seems impossible. And even if we would get this to work, visual studio code wouldn't behave as snappy as you are used to on your local computer. Therefore I propose we use [Visual Studio Code Remote SSH](https://marketplace.visualstudio.com/items?itemName=ms-vscode-remote.remote-ssh): 

> The Remote - SSH extension lets you use any remote machine with a SSH server as your development environment.

So we can open a folder on the remote machine using ssh authorization that is already in place. Save directly to the machine and use vnc to run the software.

Closing this ticket because above seems to be a better workaround.

closed 2020-05-14 by SijmenHuizenga