FS-Driverless/Formula-Student-Driverless-Simulator #18: Unreal process crashes

davidoort, 2020-05-09:
I wasn't paying attention to the machine, but after leaving the simulation running for about 7 mins I got a crash report with the following logs:
```
LoginId:860ba023ccb1a0522953411da87a6924-000003ea
EpicAccountId:

Caught signal

libUE4Editor-Core.so!FMallocBinned2::Malloc(unsigned long, unsigned int) [/home/developer/UnrealEngine/Engine/Source/Runtime/Core/Public/HAL/MallocBinned2.h:447]
libUE4Editor-Core.so!FMemory::Malloc(unsigned long, unsigned int) [/home/developer/UnrealEngine/Engine/Source/Runtime/Core/Public/HAL/FMemory.inl:21]
libUE4Editor-Renderer.so!FMeshDrawShaderBindings::CopyFrom(FMeshDrawShaderBindings const&) [/home/developer/UnrealEngine/Engine/Source/Runtime/Renderer/Private/MeshPassProcessor.cpp:656]
libUE4Editor-Renderer.so!FMeshDrawCommand::FMeshDrawCommand(FMeshDrawCommand const&) [/home/developer/UnrealEngine/Engine/Source/Runtime/Renderer/Public/MeshPassProcessor.h:450]
libUE4Editor-Renderer.so!FDynamicPassMeshDrawListContext::AddCommand(FMeshDrawCommand const&) [/home/developer/UnrealEngine/Engine/Source/Runtime/Renderer/Public/MeshPassProcessor.h:767]
libUE4Editor-Renderer.so!void FMeshPassProcessor::BuildMeshDrawCommands<TMeshProcessorShaders<FHitProxyVS, FHitProxyHS, FHitProxyDS, FHitProxyPS, FMeshMaterialShader, FMeshMaterialShader, FMeshMaterialSh$
libUE4Editor-Renderer.so!FHitProxyMeshProcessor::AddMeshBatch(FMeshBatch const&, unsigned long long, FPrimitiveSceneProxy const*, int) [/home/developer/UnrealEngine/Engine/Source/Runtime/Renderer/Private$
libUE4Editor-Renderer.so!GenerateDynamicMeshDrawCommands(FViewInfo const&, EShadingPath, EMeshPass::Type, FMeshPassProcessor*, TArray<FMeshBatchAndRelevance, TMemStackAllocator<0u> > const&, TArray<FMesh$
libUE4Editor-Renderer.so!FMeshDrawCommandPassSetupTask::AnyThreadTask() [/home/developer/UnrealEngine/Engine/Source/Runtime/Renderer/Private/MeshDrawCommands.cpp:785]
libUE4Editor-Renderer.so!TGraphTask<FMeshDrawCommandPassSetupTask>::ExecuteTask(TArray<FBaseGraphTask*, TSizedDefaultAllocator<32> >&, ENamedThreads::Type) [/home/developer/UnrealEngine/Engine/Source/Run$
libUE4Editor-Core.so!FTaskThreadAnyThread::ProcessTasks() [/home/developer/UnrealEngine/Engine/Source/Runtime/Core/Private/Async/TaskGraph.cpp:1029]
libUE4Editor-Core.so!FTaskThreadAnyThread::ProcessTasksUntilQuit(int) [/home/developer/UnrealEngine/Engine/Source/Runtime/Core/Private/Async/TaskGraph.cpp:854]
libUE4Editor-Core.so!FTaskThreadAnyThread::Run() [/home/developer/UnrealEngine/Engine/Source/Runtime/Core/Private/Async/TaskGraph.cpp:931]
libUE4Editor-Core.so!FRunnableThreadPThread::Run() [/home/developer/UnrealEngine/Engine/Source/Runtime/Core/Private/HAL/PThreadRunnableThread.cpp:25]
libUE4Editor-Core.so!FRunnableThreadPThread::_ThreadProc(void*) [/home/developer/UnrealEngine/Engine/Source/Runtime/Core/Private/HAL/PThreadRunnableThread.h:177]
libpthread.so.0!UnknownFunction(0x76da)
libc.so.6!clone(+0x3e)

```

davidoort, 2020-05-17:
This issue is almost impossible to replicate. Unreal sometimes randomly crashes but seldomly during a run. If it becomes a problem when testing the simulator for the competition we will create a new issue about reliability improvements

closed 2020-05-17 by davidoort

ugurkanates, 2021-03-22:
Sadly this exact error happens to me ( in completely unrelated airsim env)

davidoort, 2021-03-23:
@ugurkanates this is probably something to do with Unreal Engine or the AirSim lib and it's not really in the scope of this repo to fix it :)