FS-Driverless/Formula-Student-Driverless-Simulator #40: Validate that the cone sizes are in line with rules

SijmenHuizenga, 2020-05-14:
The rules describe a specific type and size of cone. We have to make sure simulated cones are the right size.

SijmenHuizenga, 2020-06-28:
Measurements in unreal engine. All units are cm.

**Small cones in the world**

Side view:
[image]

Top view:
[image]

**Large cones in the world**

Side view:
[image]

Top view:
[image]

SijmenHuizenga, 2020-07-14:
Since nobody objected above sizes, I think it is safe to assume cone dimensions are close enough to the real cones to be used in competition. Closing this ticket.

closed 2020-07-14 by SijmenHuizenga

davidoort, 2020-07-15:
Are these sizes and pictures somewhere in the docs? It might be useful for people developing vision algorithms

SijmenHuizenga, 2020-07-15:
No, they are not in the docs. These are the same sizes of cones used in all driverless formula student competitions, so I assume people know them. Could still be nice to mention it though... Maybe we should make a page with cone sizes and track details like distance between blue and yellow cones, maximum curvature, etc

Joroba3, 2023-03-05:
Perhaps it's been a while, but I just meassured our cones and there's actually quite a bit of variability in (even new) cones. Some where even 30cm tall instead of 32cm.