FS-Driverless/Formula-Student-Driverless-Simulator #45: Import ZalaZone map into repository

SijmenHuizenga, 2020-05-15:
We want to use the [zalazone maps](https://github.com/BMEAutomatedDrive/ZalaZONE-automotive-proving-ground-virtual-simulation-models). Let's get those imported in this project, setup an example track and make the car drive in here.

SijmenHuizenga, 2020-05-15:
Some time ago I gave this a try but couldn't get it to work. I raised a question [here](https://github.com/BMEAutomatedDrive/ZalaZONE-automotive-proving-ground-virtual-simulation-models/issues/1)

davidoort, 2020-05-17:
What is wrong with the current maps?

SijmenHuizenga, 2020-05-17:
We would rather use the ZalaZone maps because they are one of the big sponsors of the FS-East event.

SijmenHuizenga, 2020-05-19:
Importing the fbx file with scaling 100 makes it really big but still I only see the individual objects. Since there is no response to the github issue, I sent a dm to their facebook group.

SijmenHuizenga, 2020-05-20:
Import worked! See the linked linked ticket for tutorial. Seeing this map we have to discuss how we want to use it.

closed 2020-05-20 by SijmenHuizenga