FS-Driverless/Formula-Student-Driverless-Simulator #67: GPS: Heading

SijmenHuizenga, 2020-05-31:
Add heading to the gps information

SijmenHuizenga, 2020-06-01:
Heading information is provided in IMU so GPS does not need it. We should change `GPSYaw`  to not include Yaw.

davidoort, 2020-06-01:
GPSYaw is only used for the initial geopoint location (which I'm not sure we even need @skimslozo )

SijmenHuizenga, 2020-06-01:
Ah yes, sorry for messing those to up. For the [sensor_msgs/NavSatStatus](http://docs.ros.org/melodic/api/sensor_msgs/html/msg/NavSatFix.html) message, it seems that we are only setting latitude, longitude and altitude. Would it make more sense to use the [geometry_msgs/PointStamped Message](http://docs.ros.org/api/geometry_msgs/html/msg/PointStamped.html) message type?

davidoort, 2020-06-02:
I think we should stick to the GPS message because it is an industry standard and it has a lot of fields that we might want to use in the future. Besides, I think we're already using the covariance fields to publish the noise level of the messages. Our pipeline was going to use the navsat_transform_node described [here](https://roscon.ros.org/2015/presentations/robot_localization.pdf)

SijmenHuizenga, 2020-06-03:
Currently, this is how the message is constructed:
```
sensor_msgs::NavSatFix AirsimROSWrapper::get_gps_sensor_msg_from_airsim_geo_point(const msr::airlib::GeoPoint& geo_point) const
{
    sensor_msgs::NavSatFix gps_msg;
    gps_msg.latitude = geo_point.latitude;
    gps_msg.longitude = geo_point.longitude;
    gps_msg.altitude = geo_point.altitude;
    return gps_msg;
}
fscar_ros.gps_sensor_msg = get_gps_sensor_msg_from_airsim_geo_point(gps_location);
fscar_ros.gps_sensor_msg.header.stamp = curr_ros_time;
fscar_ros.global_gps_pub.publish(fscar_ros.gps_sensor_msg);
```
see `airsim_ros_wrapper.cpp`

So i guess currently noise levels are not being published.

davidoort, 2020-06-03:
Yes! I mentioned this to @skimslozo on Slack, but I don't know if he has gotten to it

SijmenHuizenga, 2020-06-04:
Closing in favor of #76