FS-Driverless/Formula-Student-Driverless-Simulator #79: Print warnings to console (or send them as logs to the webGUI) if the sensor suite requirements are not complied with

davidoort, 2020-06-04:
It is basically a unit test of the teams settings.json file. We might not even have to do it at runtime

smnschfr, 2020-07-17:
This would be helpful to get the exact vehicle specifications, too. Right now, I'm not sure about the final data that I can use to calculate whether my sensor placement is compliant to the official rules.

SijmenHuizenga, 2020-07-18:
@smnschfr I see your problem. Feel free to contact FSOnline if you have questions about the procedures of the competition. Here we can only answer tech-related questions. If you have a sensor configuration that you would like to be validated, feel free to open a ticket here on GitHub or send it to hello@formulastudentonline.com

SijmenHuizenga, 2020-08-11:
Closing this since FSOnline has passed.

closed 2020-08-11 by SijmenHuizenga