FS-Driverless/Formula-Student-Driverless-Simulator #86: Remove support for multi-vehicles.

SijmenHuizenga, 2020-06-05:
This project is built for formula student driverless competitions where always 1 car is on the track at the same time. We don't expect this will change within the next 5 years.

AirSim supports multiple vehicles and this makes AirSim pretty complex. Like there is this whole layer of abstraction everywhere which we just won't use in this project. This results in slower development and a higher learning curve for new engineers.

If the time comes that multiple vehicles have to be supported a new architecture has to be discussed because a single unreal engine instance will never be able to simulate sensors for multiple cars.

Over the last few weeks we have already started to remove support for multiple vehicles. The car name and topic names have been hard-coded and code for drones has been partly removed. We will continue to remove support for multiple vehicles as we go.

This ticket is here as future reference to understand why we are removing multi-vehicle support.

davidoort, 2020-06-05:
I agree that within the next 5 years FS Driverless vehicles will not be allowed to race on the same track, but I think that in sim this might begin to happen next year. I would refrain from removing this support unless it introduces a lot of complexity in some places (so far the only additional complexity I have observed is some additional for loops and passing around 1 element vectors instead of a single object) or performance issues

SijmenHuizenga, 2020-06-05:
When running multiple vehicles simultaneously, each autonomous system will have it's own ros core. A single ros node cannot connect to multiple ros cores. Thus we will need a ros bridge for each AS. Therefore I think we can remove multi-vehicle support from the ros-bridge.

You might be right that the future comes faster then we can imagine. Let's not actively remove multi-vehicle support in the AirSim plugin.

SijmenHuizenga, 2020-08-26:
Ros bridge no longer supports multiple vehicles.

closed 2020-08-26 by SijmenHuizenga