FS-Driverless/Formula-Student-Driverless-Simulator #99: Load cone locations into Unreal from file

davidoort, 2020-06-10:
Currently the simulation operator can launch the simulation .exe file from the webgui. This .exe file is fixed to a single track (currently to IM_MAP1), which means that if we wanted to run the simulation on multiple tracks during the competition, we would have to have multiple executables (releases). It would be much more convenient if we could load the cone locations into the unreal world from a file (yaml or json) so that we could have a folder with track files and based on an operator command we would load one or the other. 

Several tasks can be defined:
- Define a file structure (so that FSEast can create tracks in that format)
- Write a method inside the referee class or some other class which parses this file and returns an array poses for blue cones,  yellow cones, orange cones and misc objects such as tires, etc. 
- Modify the spline cones blueprint constructor or event script so that it calls that function and loads the appropriate meshes before anything else.

SijmenHuizenga, 2020-07-25:
For the FSOnline competition we will just be creating a map for each level manually. Selecting different can be done by adding command line arguments when running the .exe as described in the spectator documentation. So we don't need this during the fsonline 2020 competition.

However, this is still very much relevant!  Dynamically loading tracks will make random track generation possible. Also, we would no longer need to create a map for each track. SOooo yes, lets' do this someday!

s-hemer, 2020-09-01:
Reach out for the KTH guys. They already have an editor therefore. Maybe they can open-source that and you can use the same format as input ?!!!

closed 2022-08-26 by mateusz-lichota