FS-Driverless/Formula-Student-Driverless-Simulator #105: Publish tf's and odom using ENU convention

davidoort, 2020-06-12:
Teams should still write their sensor configuration in the `settings.json` file according to the NED convention. This is so that we do not have to make changes to the Unreal Lidar and Camera sensor implementations. In the ROS wrapper (where the tfs are published) we will change everything to the ENU frame (this is because ENU is the convention for ground vehicles and it is the ROS standard frame - for things like RViz - so I am sure most teams use this). This means that we will publish transforms between the `fsds/FSCar` frame and all the sensors using the ENU convention and we will publish `/fsds/testing_only/odom` and `/fsds/imu` messages in an ENU frame as well. 

What these frames look like:
[image]

davidoort, 2020-06-13:
Are the cameras in settings.json somehow parsed in a different frame than Lidar?

SijmenHuizenga, 2020-06-13:
Everything handled by airsim is in the NED frame. ROS (and in turn RViz) handle everything in ENU. In the wrapper, the settings.json is loaded in the NED frame and published in the NED frame. So the lidar that is interperted to be above the car in the airsim plugin, it is interperted to be below the car in ros.

Camera transforms are being published from a static transform publisher launched from the python script. In there I just mapped XYZ 1 to 1 from the settings.json to the transforms. When doing so I didn't think about coordinate frames so they are probably in the wrong frame.

SijmenHuizenga, 2020-06-13:
@davidoort and I had a short call about this. We discussed that for now it would be best to go back to publishing the camera transforms from the ros_wrapper node and use the original airsim transforms code (adapted to ENU) to do so.

closed 2020-06-17 by davidoort