FS-Driverless/Formula-Student-Driverless-Simulator #115: Simulator everything is bright and white

SijmenHuizenga, 2020-06-17:
[image]

When restarting the simulator somehow i got this output. Driving around works fine but everything just looks like this.

SijmenHuizenga, 2020-06-17:
After a restart i got the VK_ERROR_DEVICE_LOST error, suggesting that something was wrong with my videocard. The previous shutdown of the simulator was probably unclean. 

After closing some other programs (slack, discord, steam, chrome) and waiting for a few minutes, the problem was gone and everything worked fine again.

SijmenHuizenga, 2020-06-17:
It keeps happening. Every few times it goes white and then after a restart I get VK_ERROR_DEVICE_LOST. This is really annoying.

SijmenHuizenga, 2020-06-17:
It might be good to note that this is in the packaged windows 64x version

davidoort, 2020-06-17:
I've never observed this on the Ubuntu instances either

SijmenHuizenga, 2020-06-17:
On google cloud, using the beta-3 binaries sometimes the white thing as well. Less often than on my computer and never with the `VK_ERROR_DEVICE_LOST` error.

SijmenHuizenga, 2020-06-19:
When selecting `Package Project -> Build Configuration -> DebugGame` this problem doesn't happen.

SijmenHuizenga, 2020-06-19:
Added beta-3-debug.zip to the beta-3 release. Use that one to not have this problem.

SijmenHuizenga, 2020-07-11:
From beta-4 onward we will just be doing debug releases to not have this problem.

closed 2020-07-11 by SijmenHuizenga