FS-Driverless/Formula-Student-Driverless-Simulator #124: Weather conditions

SijmenHuizenga, 2020-06-24:
todo

davidoort, 2020-06-25:
We have agreed that the weather conditions will not change during a run and that if we manage to set the weather conditions in Unreal programmatically, we will have to define several scenarios (e.g. sunlight, rain, fog, dark) which teams can test before the competition. These teams will receive this scenario in the GO signal message

SijmenHuizenga, 2020-06-25:
A few months ago when I tried airsim for the first time I remember feeling less car grip (more slipping) when the weather was enabled. Now, looking at the code and blueprints I can't find any code that changes the vehicle dynamics or changes the ground material slipperyness when the weather is changed. My conclusion is that my memory is wrong or my experience was some kind of placebo. Currently the car behaves equal in all weather conditions.

Lidar, GPS, IMU are also unaffected by weather changes. Only the camera detects the rain drops, fog, dust and the snow.

SijmenHuizenga, 2020-08-02:
Weather and time-of-day has been programmed into the competition maps. These will be released to this repo after each competition day.

closed 2020-08-02 by SijmenHuizenga

ZYXzyx-1, 2025-03-07:
> 比赛地图中已编入天气和时间信息。比赛日结束后，这些信息将发布到此仓库中。

How did you solve this problem? This would be very helpful to me.

wouter-heerwegh, 2025-03-07:
Hi @ZYXzyx-1,

I'm not entirely sure how the weather was pre-selected when loading the maps, but you can always alter the weather by pressing F10.

ZYXzyx-1, 2025-03-12:
> Hi @wouter-heerwegh 
Yes,Thanks so much for your reply. I tried pressing the F10, but I couldn’t get the three conditions: wet road, snowy road, and leaf-covered road to show up. Also, I noticed that these weather conditions do not seem to have any physical simulation effects on the lidar or the vehicle itself. Have you looked into this issue? Really appreciate your help!

ZYXzyx-1, 2025-03-13:
It's much clearer now. Thanks a lot!