FS-Driverless/Formula-Student-Driverless-Simulator #127: LiDAR Realism and Limits

nistath, 2020-06-30:
[Reference](https://fs-online.github.io/Formula-Student-Driverless-Simulator/latest/integration-handbook)

The current rules allow for very unrealistic LiDAR setups. At the same time there are performance limitations for the number of rays that can be cast, requiring teams to be efficient with the beams they send out.
I propose the following limits for more realistic LiDAR pucks, while also allowing beams to not be wasted into the sky.

# Constraints

## Realistic LiDAR Performance Constraints
These are inspired by state of the art commercial LiDAR pucks such as the Velodyne VLS-128 and Ouster OS-2.
Realistically these limits represents a sensor most teams would be really lucky to have in real life.

| Value | Min | Max |
| - | - | - |
| Rotation Rate | 5 Hz | 20 Hz |
| # Vertical Firings for 360º | 1 | 2048 |
| # Vertical Firings per second | "0" | 20,480 |
| Minimum Vertical Inter-laser Angle | 0.18º | 180º |
| # Lasers | 1 | 128 |

## Simulation Constraints
I am not qualified to discuss the 100,000 points per second simulation performance limitation.
For context a [VLP-32](http://www.mapix.com/wp-content/uploads/2018/07/63-9378_Rev-D_ULTRA-Puck_VLP-32C_Datasheet_Web.pdf) achieves approximately 600,000 points/second in single return mode (which is what raycasting simulations can do).

We can help offset this performance limitation by allowing teams to chose the azimuth start and end angle, as well as the list of elevation angles for each laser. That can allow teams to maximize the utilization of their beams.

# Capture Realism

## Motion Correction
Can some clarification be provided on the capture frequency? At capture times do the given beams all get raycast simultaneously, or is the rotation as the vehicle moves simulated as well? To test, does a vehicle moving at 40m/s and a capture frequency equal to rotation frequency give a point cloud with concentric circles on the ground, or spirals?

## How real LiDARs capture
A real puck would shoot its lasers at a frequency of `# vertical firings per second`. For each vertical firing, an azimuth angle and a `std::array<std::pair<Distance or XYZ, Intensity>, num_lasers>` is generated and sent over UDP. The LiDAR driver takes the incoming UDP packets and converts them to point clouds. It does that either every `npackets` packets or when the azimuth crosses a `cut_angle`.
How possible is this to approximate this behavior? The concept of packets is not needed, but `npackets` would be nicer to support than a capture frequency.

SijmenHuizenga, 2020-06-30:
Thank you for this beautiful write up! :heart: 

The realistic LiDAR performance constraints seem reasonable. @lussekatter (Zsofia is an official from FSOnline) agrees so let's make a pr to get this proposal in the competition handbook. :+1:

Let's continue talking about the other suggestions! I think it has great potential but we have to figure out how we want to implement it. Let's hear some more opinions on this :smile: 

> At capture times do the given beams all get raycast simultaneously, or is the rotation as the vehicle moves simulated as well?  To test, does a vehicle moving at 40m/s and a capture frequency equal to rotation frequency give a point cloud with concentric circles on the ground, or spirals?

At this moment all rays are cast at the same time between. So no matter the speed, you will always get concentric circles on the ground. Changing this is probably possible to be implemented [here](https://github.com/FS-Online/Formula-Student-Driverless-Simulator/blob/master/UE4Project/Plugins/AirSim/Source/UnrealSensors/UnrealLidarSensor.cpp), although it requires a pretty large part of the sensor to be rewritten. However, we are limited by unreal engine ticks: if we run at 30 fps, the vehicle is moved 30 times a second and the lidar would be able to capture points at that same rate. Would that rate be enough to get the desired effect?

I'm thinking that capturing continuously (every tick) with less points a tick, might improve performance.

> How possible is this to approximate this behavior? The concept of packets is not needed, but npackets would be nicer to support than a capture frequency.

The `npackets` would definitely be possible. We have full control on how and when the lasers are sent so we could definitely support `cut_angle` and `npackets`

If we are doing this I suggest we make all of this configurable. Like that in the sensor configuration we can configure if we want simultaneous or continuous capture. When choosing continuous that we can choose to either cut-off by npackets or cut_angle. 

> For context a VLP-32 achieves approximately 600,000 points/second in single return mode (which is what raycasting simulations can do).

I am also unsure why ray-tracing is slow. [Let's see if someone knows how to improve it.](#128)

SijmenHuizenga, 2020-07-13:
Closing this since the proposed limits have been merged in #138 and a #128 has been created to investigate the lidar performance.

Feel free to open a pr if you want to implement a more advanced Lidar as described by @nistath. I think this would be a nice addition, although not necessary at this moment.

closed 2020-07-13 by SijmenHuizenga