FS-Driverless/Formula-Student-Driverless-Simulator #128: LiDAR Performance

SijmenHuizenga, 2020-06-30:
Apparently raycasting simulations can do 600,000 points per second. This simulator can only do 100,000 points per second. Does anyone have an idea on how we can improve lidar performance?

Currently lidar is using unreal's [LineTraceSingleByChannel](https://github.com/ukyan/AirSim/blob/master/Unreal/Plugins/AirSim/Source/AirBluePrintLib.cpp#L115). [Here](https://github.com/FS-Online/Formula-Student-Driverless-Simulator/blob/master/UE4Project/Plugins/AirSim/Source/UnrealSensors/UnrealLidarSensor.cpp) is the code for the lidar sensor in unreal.

llueder, 2020-08-25:
Observations:
* Our impression was that the event simulator had still trouble with 100,000 points/s, as our AS performed significantly better when we further reduced the lidars to 60,000 points/s.
* Our ubuntu instance running on a cloud VM with medium Power (6 vCPU, 2,6 GHz iirc) performed better than the event's Windows-based instance.
* (Though I don't know if our system is very sensitive or the other teams just used smaller pointclouds)

Did not investigate the nature of the problem further than to check that lidar messages come at a good rate and seem to be different, judging by the number of points in every message.
When I find some time for it after exams and stuff (certainly not within the next 40 days) and we set up a simulator again, I might try to look into it further.

chentairan, 2022-07-02:
This issue may improve its performance: https://github.com/microsoft/AirSim/pull/4505

wouter-heerwegh, 2022-07-02:
Looks great, currently it's not in an official version of Airsim yet, let's wait until it is merged and an official version is released. Changing the airsim version however is not easy, as it has been heavily cut down and modified from the original repo, but I will open a draft PR as soon as it is merged so I can start to work on getting this improvement into the sim.

There was also an issue for the camera performance microsoft/airsim#4101, this one has not progressed as much though.

wouter-heerwegh, 2022-08-16:
The PR microsoft/Airsim#4505 has been merged in the [Colosseum fork](https://github.com/CodexLabsLLC/Colosseum). I'll try and start a draft pull request to update the simulator to the newest version of Airsim (aka Colosseum).