FS-Driverless/Formula-Student-Driverless-Simulator #132: how to add Lidar sensors to settings.json

ctobasti, 2020-07-10:
I tried multiple ways to add additional Lidars to the settings.json file, but none of them were successful.
It would be helpful if you could provide a description of how they should be implemented, as well as adding parameters like HorizontalFOV... that aren`t included in the standard settings.json file, that can be found in the repository. 

Also, the Lidar data should be published under the topic /fsds/lidar/LIDAR_NAME. By changing the sensor name of a Lidar (sensor type 6) to anything else than "LidarCustom", the ros bridge prints out the following warning: "No lidar with name LidarCustom exist on vehicle".
Shouldnt the topic name adapt to the Lidar name? 

Thanks for the work on the simulation

SijmenHuizenga, 2020-07-11:
Hi! Thanks for trying the simulator! 

A detailed description of how to configure the lidar [can be found here](https://microsoft.github.io/AirSim/lidar/#lidar-configuration). You are right, we should include this in the documentation of this project as well. A full example of the lidar configuration is as follows:

```
              "LidarCustom": { 
                    "SensorType": 6,
                    "NumberOfChannels": 16,
                    "RotationsPerSecond": 10,
                    "PointsPerSecond": 100000,
                    "X": 0, "Y": 0, "Z": -1,
                    "Roll": 0, "Pitch": 0, "Yaw" : 0,
                    "VerticalFOVUpper": -15,
                    "VerticalFOVLower": -25,
                    "HorizontalFOVStart": -20,
                    "HorizontalFOVEnd": 20,
                },
```

I will try to replicate the `LidarCustom` problem later this weekend :+1: 
Where `NumberOfChannels` is the number of lasers.

May I ask out of curiosity, are you part of any formula student team?

SijmenHuizenga, 2020-07-12:
In #138 the lidar documentation is improved a lot. Would you like to check [lidar.md](https://github.com/FS-Online/Formula-Student-Driverless-Simulator/blob/lidar-reality-rules/docs/lidar.md) and [integration-handbook.md](https://github.com/FS-Online/Formula-Student-Driverless-Simulator/blob/lidar-reality-rules/docs/integration-handbook.md) and let me know if this satisfies your question?

SijmenHuizenga, 2020-07-12:
About the `No lidar with name LidarCustom exist on vehicle`: I can't reproduce this. Can you explain a bit more about your setup? Are you running on Windows or Ubuntu? Are you on a single or on multiple computers?

closed 2020-07-13 by SijmenHuizenga