FS-Driverless/Formula-Student-Driverless-Simulator #133: ROS Bridge ignores published topics not coming from localhost

smnschfr, 2020-07-11:
Hi, I've setup two nodes: the ROS Master running the Unreal simulation and the FSDS bridge (192.168.1.84) and a second node (192.168.1.42) that is communicating with this master node. The seconds node receives the data from the simulation perfectly fine. The other way around appears to work, too, at the first glance. I publish something on the second node and when I run `rostopic echo` on the master, the data is received as expected.

However, I've noticed that any `/fsds/control_command` messages only come into effect if they are published from the master node. If they come from the second node, they are completely ignored in the simulation.

To test this, I've run a [simple script](https://chaonan99.github.io/2018/rostopic-republish/) on the master that republishes the data it receives from the second node (and doesn't create any data on its own). And indeed, now the control commands take effect.

So somehow it appears to me that the bridge ignores anything not coming from localhost.

Can you recreate this issue?

Here's the `rostopic info` output when running the republish script: 

```
$ rostopic info /fsds/control_command
Type: fs_msgs/ControlCommand

Publishers:
 * /message_republish_2338_1594429949713 (http://192.168.1.84:51276/)
 * /rostopic_7921_1594429970613 (http://192.168.1.42:42637/)

Subscribers:
 * /fsds/ros_bridge (http://192.168.1.84:50857/)
 * /message_republish_2338_1594429949713 (http://192.168.1.84:51276/)

```

SijmenHuizenga, 2020-07-11:
Thanks for trying out the simulator!

When running the `FSDS bridge` there should be statistics printed to the console. Do you see a statistic being printed starting with the name `control_cmd_sub`? That one should show how many control messages the ros bridge receives. If we see no statistics on this matter, we know for sure something ros related is broken.

I will try to replicate this issue when I'm back home on Sunday.

smnschfr, 2020-07-12:
Here's the output from the FSDS bridge and the righthand terminal showing `rostopic echo /fsds/control_command`

[image]

SijmenHuizenga, 2020-07-13:
Thanks for providing those screenshots! I haven't been able to replicate this. I feel like something is wrong with your ros network configuration. In the hope to provide you with some tips I recorded a walk through of my network configuration. [You can find the video here.](https://youtu.be/Bo1WSuO_9k8)

The ros bridge is only using the ros framework to communicate with other nodes. There is no differentiation between local and remote nodes.

Let me know if you are able to solve it!

smnschfr, 2020-07-13:
Thanks for creating that video.
We're testing with local hardware and not in the Google Cloud. I haven't found any difference in the settings you used compared to ours that would explain the difference. However, the workaround we have seems to do the trick, so as long as it works natively in the event cloud environment, this should not be a big issue.

closed 2020-07-13 by SijmenHuizenga