FS-Driverless/Formula-Student-Driverless-Simulator #134: IMU Sensor Noise

wahllca, 2020-07-11:
Hi, is there any way to reduce the noise of the IMU Sensors in the Simulation?
The data below was recorded during a straight acceleration, and as you can see, the noise is quite high (~80% of scale) compared to real IMU data and therefore almost useless.

[image]

SijmenHuizenga, 2020-07-11:
@davidoort  @skimslozo what do guys you think?

skimslozo, 2020-07-11:
@wahllca The current IMU noise parameters are modelled according to a real IMU ([see datasheet](https://www.invensense.com/wp-content/uploads/2015/02/MPU-6000-Datasheet1.pdf))). How do the noise parameters of your mentioned reference IMU compare to this one?

wahllca, 2020-07-12:
Thank you very much for the quick response!
Maybe I am a little spoiled with our self-developed IMU, but you are referring to a rather cheap IMU compared to the prices of other sensors of the autonomous system. Maybe we can agree on the Sparkfun Razor IMU 9DOF (http://wiki.ros.org/razor_imu_9dof) because it is a widely used device. I could also provide a recorded rosbag for reference.

SijmenHuizenga, 2020-07-13:
I don't mind moving to a better imu model at all! It seems like a small change and I can't imagine other teams hate having access to better hardware.

@wahllca Would you be able to propose changes to [this file](https://github.com/FS-Online/Formula-Student-Driverless-Simulator/blob/master/AirSim/AirLib/include/sensors/imu/ImuSimpleParams.hpp)?

SijmenHuizenga, 2020-07-20:
@skimslozo would you be able to implement this?

skimslozo, 2020-07-24:
@SijmenHuizenga Yes, will take a look.