FS-Driverless/Formula-Student-Driverless-Simulator #135: Outlier in the Yaw Rate

wahllca, 2020-07-11:
Hey, 
sometimes the /fsds/imu publishes unrealistic high values for the yaw rate.

[image]

SijmenHuizenga, 2020-07-11:
That can't be right! Thanks for bringing this up.

davidoort, 2020-07-11:
Hi @wahllca, the IMU sensor in FSDS is not a perfect sensor so it publishes ground truth values with the addition of Gaussian noise. Even if the standard deviation of the measurement is not very high, at a publishing rate of 250Hz it is not totally out of the question to expect a value in the 3sigma or 4sigma range, which is my guess of what is happening here. I suspect that a regular filter on your autonomous system should be able to take care of this. Nonetheless, I suggest that @SijmenHuizenga checks whether these spikes occur when noise is not added to the IMU message to check if it is an issue with the ground truth

smnschfr, 2020-07-12:
If that's a spike in the Gaussian noise, shouldn't there be other, less drastic spikes visible according to the distribution?

davidoort, 2020-07-12:
That's a good point, what is the span of the x-axis in the screenshot? Can you replicate this behavior time and time again?

SijmenHuizenga, 2020-07-13:
Surprisingly there doesn't seem to be any noise in the orientation #143 So this is definitely a bug.

SijmenHuizenga, 2020-07-13:
replacted:
[image]

[image]

SijmenHuizenga, 2020-07-13:
with noise disabled: 
[image]

SijmenHuizenga, 2020-07-13:
I've done a lot of driving the last hour and tried to find a pattern. Couldn't find anything. I've looked into how the imu data is collected and it is gets it from the kinetics body in unreal. If this was an overflow somehwere in the wrapper of some sorts I would expect the outliers to be the same height.

My theory is that this data is _actually_ the observed movement of the vehicle within unreal. And something is wrong within the actual unreal simulation. Maybe somehow the vehicle trips over its self or maybe something is colliding. I have tried modifying the physics model but it's all the same. Also same behavior when using the suv car. Also tested this in a different map, same behaviour.

I don't know where the problem is. Since a workaround is available (ignore the outliers) I will stop working on this. More important tasks need my attention at this moment.

If someone else is able to figure this out I would gladly receive a pr 🥇

SijmenHuizenga, 2020-07-13:
Unmarking this as high-prio. This should be solved but I don't believe anyone is blocked by this. If this bug is bothering your autonomous system you can easily filter out the outliers.

wahllca, 2020-07-15:
While filtering the data I also noticed that the values for the yaw rate are very small. I suppose the unit is Rad/s.
With an acceleration of more than 1g and a velocity of about 6 m/s I would expect a yaw rate of about 1.6 Rad/s or am I wrong here?
[image]

wahllca, 2020-07-15:
I think I figured out the problem with the outlier. The range of the orientation is between -1 and 1 (maybe you could change it to Rad). I assume the yaw rate is obtained by deriving the orientation. When the orientation changes form 1 to -1 the derivative increases rapidly and therefore the yaw rate is implausible. 
[image]

SijmenHuizenga, 2020-07-16:
Well done figuring this out 👍 And thanks for this amazing analysis! I'm trying to solve this now

SijmenHuizenga, 2020-07-16:
There is also definitely something wrong with the scale of the angular velocity. Looking into this now.

SijmenHuizenga, 2020-07-16:
wanna see something funny?

[image]

SijmenHuizenga, 2020-07-16:
> edit: solved it!

no nvermind I just broke it soo much that we couldnt see the bug anymore

SijmenHuizenga, 2020-07-16:
Oooke we solved it, this time for real! See #154 

Thanks to @skimslozo for saving the day ❤️

SijmenHuizenga, 2020-07-16:
> The range of the orientation is between -1 and 1 (maybe you could change it to Rad)
Yes, this is because the orientation is expressed in terms of quaternions. The documentation was wrong, sorry about that.

closed 2020-07-16 by SijmenHuizenga