FS-Driverless/Formula-Student-Driverless-Simulator #139: Provide the height of the CG of the car

davidoort, 2020-07-12:
Without this information it is very hard to figure out the height of perception sensors with respect to the ground since in the settings.json a vertical offset with respect to the CG (center of gravity) of the car is provided.

SijmenHuizenga, 2020-07-13:
This is where the center of the car is. Relative to this point all sensors are configured. This center is the _center of the unreal pawn_

[image]

SijmenHuizenga, 2020-07-13:
The center of gravity is determined by the phisics asset. Each car will have the same phisics asset like so:

[image]

The dimensions of this box are width of 100cm, a length of 180cm and a hight of 50cm. This box wights 255kg and the weight is equally distributed in the box. The box is alight to to bottom of the car. So the z of the pawn position is equal to the bottom of this box. This brings the center of gravity to exactly 25 cm above the pawn location.

SijmenHuizenga, 2020-07-13:
@davidoort does above information answer your question? If this is the case I will create a pr to include this in the documentation.

davidoort, 2020-07-13:
Yes, this clarifies a lot. So the center of the car is one position and the center of gravity of the car is another. The center of the car is used for sensor positions and I assume the center of gravity is used for the dynamic model? Otherwise I'm not sure what it's function is. I'm not sure why two centers are needed and if I had to choose I would only use the center of gravity. If the sensors are positioned with respect to the center of the car then I would need the height of the origin in the first picture. I would prefer that we could just provide the position of the sensors with respect to the center of gravity in the settings.json but I don't know how difficult that is to implement...

SijmenHuizenga, 2020-07-13:
> The center of the car is used for sensor positions and I assume the center of gravity is used for the dynamic model?

Correct!

> I'm not sure why two centers are needed

This is how unreal engine works. Everything inside the world (pawns, static meshes, objects) have a position. In the case of the vehicle we have the CarPawn. The CarPawn is moved around by Unreal's physics engine that uses the weights and boundaries defined in the physics model. The center of gravity and the position of the pawn are just two different concepts, both necessary for the vehicle to drive.

> if I had to choose I would only use the center of gravity.

At this moment, nothing within AirSim uses the center of gravity. It is 100% a concept within unreal engine. I think we should just document that sensors operate relatively to the pawn center and document how the center of gravity is offset.

> If the sensors are positioned with respect to the center of the car then I would need the height of the origin in the first picture.

The height offset between the world ground and the pawn position is 0. The height offset between the world ground and the center of gravity is 25 cm.

> I would prefer that we could just provide the position of the sensors with respect to the center of gravity in the settings.json but I don't know how difficult that is to implement...

This would require a lot of changes. We would somehow have to get the CoG in the sensor code in unreal and then translate all positions everywhere. In the end, what matters is that the autonomous systems know how the position of the sensors relate to the boundaries of the car and the center of gravity. I don't see why the CoG needs to be equal to the pawn position to achieve this.

davidoort, 2020-07-13:
Thanks for your elaborate response.
> I think we should just document that sensors operate relatively to the pawn center and document how the center of gravity is offset.
This would suffice then!

PR #145 (2020-07-14): Center of gravity documentation and consistency 
Document how the center of gravity works

Moved the center of gravity of the Technion car to the center of the car. It was pretty high in the air and offset to the back. Now it is centered as described in the docs. Somehow this makes the car drive a little different. I _feel_ like it accelerates faster and slips more. I'm not too scared about changing the dynamics of the car at this moment because a change in the suspension is coming up as well and we can bundle these two changes together into a single release and announcement to the participati [... cut]
closed 2020-07-14 by SijmenHuizenga