FS-Driverless/Formula-Student-Driverless-Simulator #140: Length of cones array from unreal too high

davidoort, 2020-07-12:
So far I have just tried the Training Map, but there the following print statement in `airsim_ros_wrapper.cpp` shows a value of **19296** which is obviously incorrect.
[image]

SijmenHuizenga, 2020-07-12:
[image]

davidoort, 2020-07-12:
It is a map problem. Training map contains 19296 cones but other maps such as IM_MAP1 contains 480 cones. Renaming the issue now

davidoort, 2020-07-12:
This is issue is only a problem when running a semi-autonomous system that makes use of the `fsds/testing_only/track` topic.

smnschfr, 2020-07-12:
We've seen the same issue and were able to filter out the surplus cones by checking for location duplicates. After doing that, the map is correct.

SijmenHuizenga, 2020-07-12:
Cool! I will fix the training map monday.

SijmenHuizenga, 2020-07-13:
Progress? 
[image]

SijmenHuizenga, 2020-07-13:
> The Construction Script is designed to execute upon creation of the Blueprint object, and **again whenever that object is updated in some way.** ([source](https://docs.unrealengine.com/en-US/Resources/ContentExamples/Blueprints/1_4/index.html))

In the spline_cones we were using the construction script to create the cones and added them to an array. When the object was updated all cones were created again. After adding a clear of the arrays each update it works

[image]

[image]

closed 2020-07-13 by SijmenHuizenga