FS-Driverless/Formula-Student-Driverless-Simulator #144: Offset between cone positions in the ground truth topic and Unreal Engine

davidoort, 2020-07-13:
[image]

SijmenHuizenga, 2020-07-13:
Which commit is this?

SijmenHuizenga, 2020-07-14:
Is it coincidence that the offset looks like to be the width of the vehicle?

SijmenHuizenga, 2020-07-14:
@davidoort Do you have an idea how we can test this without the FSTD autonomous system? I would like to confirm this is a FSDS. In the picture you show it could also be an FSTD integration problem.

davidoort, 2020-07-14:
@SijmenHuizenga the offset seems to only occur in the x direction (i.e. the direction the car is pointing to at the start). This indicates that if there is a problem, it could do with the initial x car position in unreal (or how that position is defined vs how we interpret it to be defined). In Referee.cpp we load this start_pos (which comes from the Refereeblueprint) which is subtracted from all the cone positions. I interpret that position as the CG of the car, but maybe it is another weird reference point on the vehicle.
[image]

>  Do you have an idea how we can test this without the FSTD autonomous system? I would like to confirm this is a FSDS. In the picture you show it could also be an FSTD integration problem.

I can't think of an easy way that doesn't require creating a visualization publisher and a map to car  frame broadcaster based on odometry.

It would be really helpful if for now you could clarify what this start pos actually is

smnschfr, 2020-07-21:
An x-offset of 2 meters seems to match pretty well.
This was just obtained through trying out, so no idea where the error lies.

SijmenHuizenga, 2020-07-24:
I found a way to test it easely:

1. Create a map in unreal engine with only 4 cones

2. Add the following debug lines to the wrapper:

```
diff --git a/ros/src/fsds_ros_bridge/src/airsim_ros_wrapper.cpp b/ros/src/fsds_ros_bridge/src/airsim_ros_wrapper.cpp
index 80dcd5c..5d764a2 100644
--- a/ros/src/fsds_ros_bridge/src/airsim_ros_wrapper.cpp
+++ b/ros/src/fsds_ros_bridge/src/airsim_ros_wrapper.cpp
@@ -83,6 +83,9 @@ void AirsimROSWrapper::publish_track() {
     // Get car initial position
     car_start_pos = state.car_start_location;

+^M
+    std::cout << "car" << " x: " << car_start_pos.x << " y: " << car_start_pos.y << std::endl;^M
+^M
     fs_msgs::Track track;
     for (const auto& cone : state.cones) {
         fs_msgs::Cone cone_object;
@@ -90,6 +93,7 @@ void AirsimROSWrapper::publish_track() {
         cone_object.location.x = cone.location.x != 0 ? (cone.location.x - car_start_pos.x)*0.01 : 0;
         // Negative sign follows ENU convention
         cone_object.location.y = cone.location.y != 0 ? -(cone.location.y - car_start_pos.y)*0.01 : 0;
+        std::cout << "cone" << " x: " << cone_object.location.x << " y: " << cone_object.location.y << " color: " << cone.color << std::endl;^M
         if (cone.color == CarApiBase::ConeColor::Yellow) {
             cone_object.color = fs_msgs::Cone::YELLOW;
         } else if (cone.color == CarApiBase::ConeColor::Blue) {
```

3. Use the measuring tool in unreal editor to check the distance between car and cones
4. profit

davidoort, 2020-08-02:
According to recent tests, this issue does not seem to have been fixed in either map. Is there a plan to fix this before competition?
[image]
[image]

SijmenHuizenga, 2020-08-02:
When testing on the trainingmap with the following cones:

[image]

So the cones relative to the car should be at x 284

[image]

However, they are at 4.8 (using the above print statements output the following)

```
car x: 4575.15 y: 8577.82
cone x: 4.82257 y: -2.06277 color: 0
cone x: 4.82257 y: 1.43723 color: 1
```

So yea you are right, the problem still exists.

davidoort, 2020-08-02:
Thanks for creating this test setup! We can either find the root source of the issue in Unreal Engine or we can simply correct for this known offset (198.257 cm) in the ROS wrapper (which at least would make it ready by the competition)

smnschfr, 2020-08-02:
Please let us know if you change anything here before the competion. Right now, we have implemented a correction for this offset in our AS which would need to be reversed once it's fixed in the signal.

SijmenHuizenga, 2020-08-05:
Apparently the vehicle is spawned offset from the playerstart. 

The arrows show the playerstart component, the vehicle is untouched.
[image]

The offset is probably caused by `void ASimModeBase::setupVehiclesAndCamera()`. This all is because airsim supports changing the spawn position from settings.jon.

mateusz-lichota, 2022-08-24:
This was fixed in #274

commit adb313cb (2020-07-24): Fix #144 by using the ground truth cone positions instead of the translations of the cones in the cone spline
closed 2022-08-24 by mateusz-lichota