FS-Driverless/Formula-Student-Driverless-Simulator #155: Camera info not available?

tomasr8, 2020-07-16:
I'm wondering if there is a way to obtain the camera info from the ros bridge. There is a topic called `/camera/camera_info`, but `rostopic echo` reveals that it doesn't publish any data? Am I doing something wrong? How can I get the calibration matrix?

Thank you ;)

SijmenHuizenga, 2020-07-16:
You are not doing anything wrong. Indeed, currently `/camera/camera_info` is not available.

> How can I get the calibration matrix?

When you configure the parameters in your `settings.json` you select the resolution, position, rotation and field of view of the camera. What more information do you need for your calibration matrix?

tomasr8, 2020-07-16:
Thanks for the reply! Looks like the information in `settings.json` is all we need.

SijmenHuizenga, 2020-07-16:
Nice! 

Oh and just so you know, the transforms between all sensors, including the camera's, are published on `/tf` as well 👍

closed 2020-07-16 by SijmenHuizenga