FS-Driverless/Formula-Student-Driverless-Simulator #158: Can't open UE4 project

Dokans, 2020-07-17:
When trying to open UE4 project, I got system critical error and couldn't open it. UE4 genereated these 2 log files. Everything was tried on release-4. Prior to opening UE4 project I did everything acording to [Setup your development environment](https://fs-online.github.io/Formula-Student-Driverless-Simulator/beta-4/get-ready-to-develop/) and [How to develop the FSDS project](https://fs-online.github.io/Formula-Student-Driverless-Simulator/beta-4/how-to-develop/). All of this was done on fresh install of Windows 10.

[FSOnline_excerpt.log](https://github.com/FS-Online/Formula-Student-Driverless-Simulator/files/4939303/FSOnline_excerpt.log)
[CrashReportClient_2.log](https://github.com/FS-Online/Formula-Student-Driverless-Simulator/files/4939304/CrashReportClient_2.log)

SijmenHuizenga, 2020-07-18:
Can you check the content of `Users/eForce Driverless/Formula-Student-Driverless-Simulator/UE4Project/Plugins/AirSim/Content/AirsimSpectatorPawn.uasset`? I expect that the repository has been cloned without LFS enabled and thus the file contains some git reference key instead of the expected binary content.

SijmenHuizenga, 2020-07-20:
Closing this because no response, if you still can't open it, I welcome you to respond and we can re-open :)

closed 2020-07-20 by SijmenHuizenga

llueder, 2020-07-31:
FYI and future reference for any other guys who come late to the party
I encountered the same error and had in fact missed git lfs.
apt install git-lfs does not suffice, one must run `git lfs install`, or do a manual `git lfs pull` in the repo.