FS-Driverless/Formula-Student-Driverless-Simulator #177: Using ground truth topics at FSOnline competition

davidoort, 2020-07-24:
I was wondering if (at least in the first edition of FSOnline this year) teams could be allowed to use the ground truth topics `/groundtruth/testing_only/odom` and `/groundtruth/testing_only/track` at the cost of a penalty that would only allow them to rank higher than teams using the same or higher number of groundtruth topics. This might help teams who are new to the simulator and haven't had time to integrate their perception and state estimation pipeline but still want to race

SijmenHuizenga, 2020-07-24:
The final decision on this topic is up to the FSOnline officials because it requires a rulebook change. I guess FSOnline will be more inclined to allow this if multiple teams think this is a good idea.

Personally I have no opinion on this matter.

Plavit, 2020-07-25:
eForce Driverless would appreciate having this as a backup option 👍

SijmenHuizenga, 2020-07-26:
I suggest you two send an email to hello@formulastudentonline.com with this proposal. That is the official channel for rule change requests.

closed 2020-07-26 by SijmenHuizenga