FS-Driverless/Formula-Student-Driverless-Simulator #184: AirSim Deployment on Google Cloud

smnschfr, 2020-07-25:
Hi team,

to my information, testing days for FS Online are planned to be from 3rd to 5th of August. 
Due to time constraints (exams) on these days, I'd prefer to test the cloud setup earlier.
I could imagine you have scripts for deploying the AirSim + bridge on Google Cloud. If yes, could you make these available so that we could perform integration tests a few days earlier on our own instance?

Thanks and best regards
Simon

SijmenHuizenga, 2020-07-25:
Hi Simon,

Thanks for asking these questions! Great to hear you are eager to participate.

Too bad your exams overlap. Maybe someone else in your team can be there during the tests?

Running the simulator involves quite some manual steps, including launching a windows google cloud instance. Unfortunately launching the stack is not fully automated at this moment. I don't see an easy way of providing you with a support-free testing setup.

An 'easy' way of doing an integration test could be to record a rosbag with sensordata+signals and play the whole thing to see if your autonomous system keeps running like it should.

Just for my overview, which team are you part?

Greetings,
Sijmen

SijmenHuizenga, 2020-07-25:
I'm not that eager to move testdays because the planning is already quite tight. Unless all teams have this problem of overlapping exams. I will leave this ticket open for a few days to see if that is the case.

closed 2020-08-11 by SijmenHuizenga