FS-Driverless/Formula-Student-Driverless-Simulator #185: Preloading of maps in the FSOnline event

smnschfr, 2020-07-25:
In the [docs](https://github.com/FS-Online/Formula-Student-Driverless-Simulator/blob/master/docs/integration-handbook.md), the "Starting" section explains how the AS has to deal with different tracks and disciplines. 
Does this part need to be automated, or can we simply wait for the information to be communicated by race control ("up next: track B autocross") , login via SSH and run the appropriate launchfiles?

SijmenHuizenga, 2020-07-25:
On every competition day, the team's remote access to the instances is blocked and restored after the events are finished. During the day, between the events, you won't be able to access your autonomous system So yes, your autonomous system must be able to automatically switch between autocross and trackdrive.

SijmenHuizenga, 2020-07-25:
If you have any more questions, feel free to re-open :+1:

closed 2020-07-25 by SijmenHuizenga

smnschfr, 2020-07-26:
After sending /fsds/signal/finished, can we assume that /fsds/signal/go will be stopped soon*? 

*like within 60 seconds