FS-Driverless/Formula-Student-Driverless-Simulator #189: Custom vehicle models integration during competition

Plavit, 2020-07-27:
Following up to #47 .

@SijmenHuizenga I wonder - is there a plan to use custom models for the competition? If yes, how/in what format will we be providing the models? 

ueassets / pawns maybe?

SijmenHuizenga, 2020-07-27:
Hi Marek! Teams can provide their 3d vehicle models as an unreal pawn as described [here](https://fs-online.github.io/Formula-Student-Driverless-Simulator/latest/import-car-3d-model/). If you want the 3d model to be public you can create a pr to this repo. If you want it to be kept private you can email it to me at sijmen.huizenga@formulastudentonline.com. These will be multiple uasset files. 

Adding your own 3d model is quite tedious. I had a lot of trouble with writing that tutorial and I haven't gotten my teams vehicle to drive. So don't feel bad if you can't get it to work straight away ;)

Plavit, 2020-07-27:
@SijmenHuizenga yeah, we are still only trying to get it work...

Thanks for the answer, could you please add this information about sharing the pawn to the end of the tutorial? I looked there before, but no info about sharing is stated there...

Thanks!
Marek

SijmenHuizenga, 2020-07-27:
Good idea! I have created a pr. Would you be able to review it @Plavit ?

Plavit, 2020-07-27:
@SijmenHuizenga Made some minor suggestions, otherwise solved, thanks a lot Sijmen!

closed 2020-07-27 by Plavit

GuihVicentini, 2022-05-06:
Hi Guys, I know this is already closed, but could anyone help me to add my custom model to the simulator?
I've followed all the tutorials but so far I only got it to be spawn and I am able to see the animation when I press arrow keys. But the car doesn't move.

The message is: Collision#32 with floor - ObjID 39

Thanks for the help!

wouter-heerwegh, 2022-05-06:
Hi @GuihVicentini,

I've never done this before but I think the car is just spawning too low. Is your car a lot higher? Is there maybe a way to move your actor reference point downwards?

I think that if you press backspace it will spawn higher up, but if your actor reference point is too high up, then it will spawn in the ground.

It would also maybe be nice to share your model, so i can have a look

GuihVicentini, 2022-05-09:
Hi @wouter-heerwegh thanks a lot for the fast response :)

I was able to make it work! I basically followed [this video](https://youtu.be/BKbKfF0jTLY), specially the UE part. In blender I followed your tutorial in the doc pages, but what was also important was the scale.

I guess what was really happening was that the wheels where not getting the mesh colliders or the radius was too small. So the animation was playing but the car was not moving.

Anyhow thanks a lot! :)