FS-Driverless/Formula-Student-Driverless-Simulator #196: Problem with launching 1.1.0 released version

KarasevPavel, 2020-07-29:
Hello! I has tried to run an FSDS.exe as was mentioned in Getting Started instruction https://fs-online.github.io/Formula-Student-Driverless-Simulator/latest/getting-started/, but I had only a black screen with a title in the left upper corner of the screen, as on the screenshot. I am using Windows 10 Pro with Formula-Student-Driverless-Simulator repository installed in my Windows home directory.

[image]

KarasevPavel, 2020-07-29:
Solved! I have installed a GeForce Game Ready Driver for my GTX1650 GPU of version 451.67 (latest version) and problem was solved.

closed 2020-07-29 by KarasevPavel

SijmenHuizenga, 2020-07-29:
Welcome to the project 👋 

Glad you solved it!