FS-Driverless/Formula-Student-Driverless-Simulator #203: Extreme collisions with formulas flying

Plavit, 2020-08-01:
While testing around, we have encountered what is potentially a really nasty bug.

During driving, the vehicle appears to have crashed into some invisible ninja wall which sent it flying into a different direction - completely messing up all algorithms. See video for reference:
https://youtu.be/lBl9r7biYxE

Static picture of the incident:
[image]

This could be caused by a collision with a cone, we are not sure.

Is there a plan to make sure this does not repeat at comp? What is the expected FSO procedure if this happens and ruins a team's attempt?

SijmenHuizenga, 2020-08-01:
Flying off already, amazing! 

No but seriously, this must be fixed. I will look into it. Thanks for the report and the video!!!

If this happens during competition a re-run will be issued. This is dictated by the following rule in [the rulebook under section 3.1](https://fso-srv.sze.hu/wp-content/uploads/2020/06/FS-Online-2020-Virtual-Driverless-Event-Rules_vol2.pdf):

> In case the simulator does not perform as expected during a run, a rerun can be requested by the officials.

SijmenHuizenga, 2020-08-01:
It looks like the track is not exactly straight:

[image]

Plavit, 2020-08-01:
@SijmenHuizenga Do you mean flat?

Yes, it is not flat when the formula comes from the light gray asphalt onto the dark gray asphalt - but this is not a big issue, we thought it was a feature tbh 😅

However the bug we described above seems to happen on a flat surface as well

SijmenHuizenga, 2020-08-01:
Yes I mean flat, sorry for my Dutchy English ;)

Yes, the pavement being raised a bit is a feature. However, the light gray asphalt within itself should be flat. Because the pavement is made up of individual pieces linked together. If they do so at an angle weird behavior happens:

[image]

However... After some more testing I can conclude that this does not solve the problem. Back to debugging 😅

SijmenHuizenga, 2020-08-01:
Soooo I havn't been able to find it. So far I have ruled out the following things:

* It has nothing to do with cones because it also happens in spots without cones
* It has nothing to do with the spectator trigger lines because it happens in spots far away from any spectator trigger
* The track if as flat as it can be, so no problems there

Has anyone been able to replicate this on the dark gray asphalt? Maybe it has something to do with the light-gray asphalt?

SijmenHuizenga, 2020-08-01:
I have disabled collision on the light-gray pavement. This will be included in the next version.

closed 2020-08-02 by SijmenHuizenga

Plavit, 2020-08-02:
@SijmenHuizenga Cool, thanks for the fix! Maybe it was invisible nijas after all...

If the colisions on the light-gray pavements are off, will this not interfere with cone hitting? And will the "pavement raise bump" be still there?

SijmenHuizenga, 2020-08-02:
> If the collisions on the light-gray pavements are off, will this not interfere with cone hitting?

Cones work just like before. Nothing has changes. You will be able to test in version 1.2.0 that will be released in the next hour.

>  will the "pavement raise bump" be still there?

Unfortunately not. But I think keeping the bump is way less important then the invisible ninja so it's gone.

Plavit, 2020-08-02:
@SijmenHuizenga Yes agreed, thanks for the information and fix 👍

PR #206 (2020-08-02): FSOnline competition maps preperation
Added FSOnline adverts into the competition maps, fix #203, added all maps into main menu
commit b0d8a9e7 (2020-08-02): Added FSOnline adverts into the competition maps, fix #203, added all maps into main menu