FS-Driverless/Formula-Student-Driverless-Simulator #224: Can't get simulator to start

wouterheerwegh, 2020-08-11:
Hi,

I'm trying to get the simulator working with our autonomous system but I can't seem to load the simulator. I downloaded the release 1.3.0 and tried to run the FSDS.exe file. It loads like this: 
[image]

Whenever I try to run the simulator it crashes with a fatal error except for when I choose one of the competition maps.
I've also tried building from source as suggested in the documentation. AirSim seems to build correctly as no errors are given. If I load the project in unreal engine 4.25.3 I get this error:

```
Missing cached shader map for material DefaultMaterial, compiling. 
The file 'C:/Users/wouter/Desktop/Formula-Student-Driverless-Simulator-1.3.0/UE4Project/Plugins/AirSim/Content/Blueprints/AirsimSpectatorPawn.uasset' contains unrecognizable data, check that it is of the expected type.
The file 'C:/Users/wouter/Desktop/Formula-Student-Driverless-Simulator-1.3.0/UE4Project/Plugins/AirSim/Content/Blueprints/AirsimSpectatorPawn.uasset' contains unrecognizable data, check that it is of the expected type.
Failed to find object 'Class /AirSim/Blueprints/AirsimSpectatorPawn.AirsimSpectatorPawn_C'
CDO Constructor (AirSimGameMode): Failed to find /AirSim/Blueprints/AirsimSpectatorPawn.AirsimSpectatorPawn_C

The file 'C:/Users/wouter/Desktop/Formula-Student-Driverless-Simulator-1.3.0/UE4Project/Plugins/AirSim/Content/Blueprints/BP_PIPCamera.uasset' contains unrecognizable data, check that it is of the expected type.
The file 'C:/Users/wouter/Desktop/Formula-Student-Driverless-Simulator-1.3.0/UE4Project/Plugins/AirSim/Content/Blueprints/BP_PIPCamera.uasset' contains unrecognizable data, check that it is of the expected type.
Failed to find object 'Class /AirSim/Blueprints/BP_PIPCamera.BP_PIPCamera_C'
CDO Constructor (CarPawn): Failed to find /AirSim/Blueprints/BP_PIPCamera.BP_PIPCamera_C

The file 'C:/Users/wouter/Desktop/Formula-Student-Driverless-Simulator-1.3.0/UE4Project/Plugins/AirSim/Content/VehicleAdv/WheelData/Slippery.uasset' contains unrecognizable data, check that it is of the expected type.
The file 'C:/Users/wouter/Desktop/Formula-Student-Driverless-Simulator-1.3.0/UE4Project/Plugins/AirSim/Content/VehicleAdv/WheelData/Slippery.uasset' contains unrecognizable data, check that it is of the expected type.
Failed to find object 'Object /AirSim/VehicleAdv/WheelData/Slippery.Slippery'
```
Does anyone have an idea?

Kind regards,
Wouter from Formula Electric Belgium

SijmenHuizenga, 2020-08-11:
Hi Wouter, thanks for reporting this issue! I will look into it today

SijmenHuizenga, 2020-08-11:
> Whenever I try to run the simulator it crashes with a fatal error

I tried to run the 1.3.0 version on my computer ant it doesn't crash. What is the fatal error you get?

> If I load the project in unreal engine 4.25.3 I get this error

Can you check the content of `C:/Users/wouter/Desktop/Formula-Student-Driverless-Simulator-1.3.0/UE4Project/Plugins/AirSim/Content/Blueprints/AirsimSpectatorPawn.uasset`? I expect that the repository has been cloned without LFS enabled and thus the file contains some git reference key instead of the expected binary content. To fix this, see https://github.com/FS-Online/Formula-Student-Driverless-Simulator/issues/158#issuecomment-667276343

wouterheerwegh, 2020-08-11:
When I run the binary and press run simulation on with a training map I get this popup Fatal error!
[image]

If I run a competition map I get this:
[image]

>@SijmenHuizenga: Can you check the content of `C:/Users/wouter/Desktop/Formula-Student-Driverless-Simulator-1.3.0/UE4Project/Plugins/AirSim/Content/Blueprints/AirsimSpectatorPawn.uasset?`

The file was indeed not downloaded (as it was part of lfs), I just downloaded the source code of the 1.3.0 release on the release page. Cloned the repository in a different directory and the files got pulled just fine. Unreal engine managed to load and after putting the repository in my home directory I was able to run the simulation

wouterheerwegh, 2020-08-11:
The binary also works perfectly when I put its directory in my home folder, thanks for the help :)

SijmenHuizenga, 2020-08-11:
It turns out that in version 1.3.0, selecting one of the competition maps indeed crash. Previously I was testing for a crash on a competition testing maps. Turns out, the 1.3.0 version does not include the competition because they were still secret at the time. The dropdown already includes those maps because we were using those to test. I'm so sorry for not documenting this properly and not thinking of this this earlier today.

Somewhere tonight I will publish 1.4.0: a new version with all the competition maps included.

SijmenHuizenga, 2020-08-11:
**Error on running the training map**

I suspect that you did not have a settings.json in the correct place. After #148 the error message should have been something along the lines of "settings.json file does not exist". Testing to run 1.3.0 without settings.json gives me the same fatal error you ran into. So I am re-opening #90 to fix this.

wouterheerwegh, 2020-08-11:
Thanks for the help and the quick responses, I'll use the testing maps for now :)

closed 2020-08-11 by SijmenHuizenga