FS-Driverless/Formula-Student-Driverless-Simulator #227: Changing ROS_HOSTNAME

wouterheerwegh, 2020-08-13:
Hi,

I was trying to run the simulator on my desktop together with the ros bridge and control the car by running the joystick node on a laptop (to check if i was publishing on the same ros core). But it doesn't seem to be connected to the simulator anymore, as in, the ros bridge crashes because it doesn't get camera data.

Things I did were:
export ROS_HOSTNAME=myip
export ROS_MASTER_URI=http://myip:11311
running the ros bridge with host:=myip

Does the simulator need to be compiled again if i want to change the ROS_HOSTNAME?

Kind regards,
Wouter from Formula Electric Belgium

smnschfr, 2020-08-13:
Not sure if that makes a difference here, but generally it is advised to use ROS_IP instead of ROS_HOSTNAME if you're using an IP address.

When you're using two systems, the setup should be as follows:
System 1 running the roscore:
export ROS_IP=system1ip

System 2 not running the roscore:
export ROS_IP=system2ip
export ROS_MASTER_URI=http://system1ip:11311

Not sure from the description whether it was already setup this way.
There's no need to recompile anything, but after changing such settings, you should restart all ROS node including the core and make sure the new settings are sourced in each terminal.

SijmenHuizenga, 2020-08-13:
> the ros bridge crashes because it doesn't get camera data

Can you share the logs running up to and during the crash?

wouterheerwegh, 2020-08-13:
Hi @SijmenHuizenga,

Using ROS_IP instead of ROS_HOSTNAME worked immediately. When i start the joystick node on my laptop, it sends data to my desktop (running the simulator) and I see changing data on the topic `/fsds/control_command` but the car doesn't move. Starting the ros bridge in manual mode does allow me to move the car with the keyboard.

Am I missing something?

Do you still want the logs of the crash? I've been looking into them and there is not really something interesting, the camera node just signals a SIGTERM after logging 'spin'.

SijmenHuizenga, 2020-08-13:
So, your setup is as follows:

* the simulator, ros bridge and ros master is running on your desktop 
* the joystick is running on your laptop connected to the ros master on your desktop using ROS_IP and ROS_MASTER_URI.
* the messages published by your laptop are received on your desktop but the ros bridge does receive those messages?

About  the logs: Thanks for sharing! It's weird indeed, here is nothing in there telling us what went wrong...

wouterheerwegh, 2020-08-13:
That's indeed my setup, I ```rostopic echo /fsds/control_command``` on my desktop.

Logs: It also doesn't say that it did connect to the simulator, but it didn't give an error message that it couldn't connect to it.

smnschfr, 2020-08-13:
Sounds like it may be the same problem we had described in https://github.com/FS-Driverless/Formula-Student-Driverless-Simulator/issues/133

I assumed it was our setup and not a general issue and did not follow up on this further, but we're still using the republishing script.

SijmenHuizenga, 2020-08-13:
Hmm this is magical. I really want to see this with my own eyes. @wouterheerwegh are you up for a short call where you share your screen and show the problem? I will be [https://meet.google.com/iai-rnrf-dex](here) for the next hour or so.

SijmenHuizenga, 2020-08-13:
Thanks to @wouterheerwegh we now have a reproducible testcase:

1. Start ros core on laptop (ubuntu)
2. Start `rostopic pub -r 10 /fsds/control_command` on laptop (ubuntu)
3. Start `rostopic echo /fsds/control_command` op desktop (Windows WSL Ubuntu). Now the messages are being received
4. Start the ros bridge on desktop (Windows WSL Windows). Messages are not received by bridge, still incomming to echo on desktop
5. Restart `rostopic echo` on desktop (windows WSL ubuntu). Now no more messages are received.
6. Run `roswtf` on desktop (windows wsl windows) and see the following error:
[image]

So somehow the ros bridge kills the node...? It's a kind of magic

SijmenHuizenga, 2020-08-13:
This seems to be the same problem I encountered in #199.

wouterheerwegh, 2020-08-13:
What ended up working was:

- Running the simulator on my desktop (pc1)
- Running the bridge, master, and other stuff on my laptop (pc2)
- The brigde is run with the argument host:=pc1ip : `roslaunch fsds_ros_bridge fsds_ros_bridge.launch host:=pc1ip`
- Cameras were removed from settings.json (caused timeouts)

Conclusion: WSL seems to do some weird things with tcp ports.

During this test, the firewall of pc1 was completely disabled.
Both ```rostopic pub -r 10 /fsds/control_command fs_msgs/ControlCommand "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: ''
throttle: 1.0
steering: 0.0
brake: 0.0" ``` and the joystick node (`roslaunch joystick joystick.launch` worked perfectly.

SijmenHuizenga, 2020-08-13:
Unless someone else is able to figure out what the root cause is, I think the best thing we can do is stop recommending the use of WSL in the docs and add a note about this issue... 

Let's leave this open so that maybe some ROS networking wizard finds it and can help out.

wouterheerwegh, 2020-08-13:
I modified the subscriber for the `/fsds/control_command` topic to force a UDP connection instead of a TCP connection. Seems to work perfectly with ROS master on pc1 (see earlier messages).

Joystick works and cameras aswell.

Pull request #229

SijmenHuizenga, 2020-08-14:
Thanks for testing that workaround @wouterheerwegh! As commented in the PR, I think we should first investigate more to see if we can find the root cause before we leap to udp.

**Idea 1**: We know `rostopic echo` from wsl _does_ work fine. Subscribing to a topic from within _can work_, bot somehow for the ros bridge _it doesn't_. So maybe we are doing something wrong within the ros bridge. We could investigate this by removing code until we find the smallest piece of code that reproduces the problem. Then we can also post a question to the ros forum for help.

**Idea 2**: In this issue and in #199 we have seen that WSL1 is doing some weird magic with ros networking. We could try [WSL2](https://jack-kawell.com/2020/06/12/ros-wsl2/): 
> WSL2 is a complete reworking of WSL that shifts from what was essentially an emulation based approach to a virtualization based approach. WSL2 comes with a full Linux kernel and runs as a very lightweight VM on top of hypervisor.

Maybe the problem disappears and we can recommend everyone to use WSL2 over WSL1?

wouterheerwegh, 2020-08-14:
@SijmenHuizenga I'll first try to run the basic subsriber and publisher as described in the ros documentation (just sending some text messages), if even this wont work, we know we don't have to look into the code of this repo

wouterheerwegh, 2020-08-19:
So I've tested a couple of things:

- python scripts publishing the data with the UDP connection as suggested in pull request #229 &rarr; didn't work
- Used the most basic subscriber and publisher as described in the ROS documentation ([here](http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28c%2B%2B%29)) both in C++ and python &rarr; didn't work
- Changed to WSL2 &rarr; WSL2 uses a simulated subnet, meaning I would have to port forward a connection to my LAN which does work if I want to ping the WSL instance but when trying to connect to the ROS master doesn't work.

To give a bit more context, my desktop (pc1) has IP 192.168.1.103, the WSL instance has IP 172.27.245.16, I can port forward a connection from my desktop to the WSL instance for instance 192.168.1.103:3390 is connected to the WSL IP. However, trying to change the 'ROS_MASTER_URI' on my laptop (IP 192.168.1.105) to 'http://192.168.1.103:3390' didn't change much as ROS checks for both the host and port arguments. I can't set it to the WSL IP, because then it won't be able to find the ROS master, and if I change it to 'http://192.168.1.103:3390' it will fail to connect because the host IP doesn't match.

At the moment I don't really see a better solution than to just publish the ```fsds/control_command``` through a C++ node and by forcing a UDP connection on the simulators side (see pull request #229).

closed 2022-08-15 by wouter-heerwegh