FS-Driverless/Formula-Student-Driverless-Simulator #246: ROS2 bridge implementation

wouterheerwegh, 2020-08-30:
As ROS2 is coming along nicely, it would be nice to have a ros2 bridge aswell.

ROS2 is supposed to be faster than ROS1 but for instance launch files are not implemented yet.
You can follow their progress [here](http://design.ros2.org/articles/changes.html)

mateusz-lichota, 2021-11-24:
ROS2 is now quite mature, with a large majority of packages being available on ros-galactic. Can I get a brief on what it would take to port the bridge to ROS2? I'd be happy to perform the porting myself and submit a pull request.

wouter-heerwegh, 2021-11-24:
Hi @mateusz-lichota,

Normally only the files in the ros folder should be converted to ros 2. If you maje a draft pull request I can try to help a bit.

mateusz-lichota, 2021-11-25:
All right. Before I start the work though, how will that be incorporated into the project? I assume that the ROS1 bridge will have to be kept for compability with existing software. I see at least two ways of going about this:

1. **Creating a separate branch for ROS2 integration**. I think this would unnecesarily complicate further development, as all changes would have to be synchronized between the branches
2. **Creating an additional `ros2` directory next to the `ros` directory**, with similar structure and all packages renamed to indicate that they're ROS2 compatible, e.g. `fsds_ros2_bridge`. I think this option is better, but I'm not familiar enough with this project to make that call.

Let me know what you think @wouter-heerwegh

wouter-heerwegh, 2021-11-25:
I think option 2 is indeed the best, making sure all branches woul have the same features would be a pain. That's why I have been holding of on changing all the docs to noetic. I think that at some point, I will just make a branch for melodic and that version will then be deprecated.

mateusz-lichota, 2022-01-11:
Took a while, but here it is: #279

GuihVicentini, 2022-03-10:
Hi Guys, I'm trying to run the ros2 bridge, but I keep getting the following error:

ros2 launch fsds_ros2_bridge fsds_ros2_bridge.launch.py host:=192.168.1.170
[INFO] [launch]: All log files can be found below /home/user/.ros/log/2022-03-10-22-19-36-847627-user-VirtualBox-68476
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [fsds_ros2_bridge_camera-1]: process started with pid [68478]
[INFO] [fsds_ros2_bridge_camera-2]: process started with pid [68480]
[INFO] [fsds_ros2_bridge-3]: process started with pid [68482]
[fsds_ros2_bridge-3] [INFO] [1646947177.119312079] [ros_bridge]: Manual mode: 0
[fsds_ros2_bridge-3] Ground Speed Sensor
[fsds_ros2_bridge-3] Gps
[fsds_ros2_bridge-3] Imu
[fsds_ros2_bridge-3] Lidar
[fsds_ros2_bridge-3] Lidar
**[fsds_ros2_bridge-3] Waiting for connection - 
[fsds_ros2_bridge_camera-2] Waiting for connection - 
[fsds_ros2_bridge_camera-1] Waiting for connection -** 
[fsds_ros2_bridge-3] terminate called after throwing an instance of 'std::runtime_error'
[fsds_ros2_bridge-3]   what():  Failed connecting to rps client (airsim). Is the simulator running?
**[fsds_ros2_bridge_camera-1] Connected to the simulator!
[fsds_ros2_bridge_camera-2] Connected to the simulator!
[ERROR] [fsds_ros2_bridge-3]: process has died [pid 68482, exit code -6,** cmd '/home/user/Formula-Student-Driverless-Simulator/ros2/install/fsds_ros2_bridge/lib/fsds_ros2_bridge/fsds_ros2_bridge --ros-args -r __node:=ros_bridge --params-file /tmp/launch_params_51mvgty4 --params-file /tmp/launch_params_fx8_d9ra --params-file /tmp/launch_params_4vexol6u --params-file /tmp/launch_params_a99h5mzf --params-file /tmp/launch_params_ea_jj4dd --params-file /tmp/launch_params_x50_y21n --params-file /tmp/launch_params_a8n8pkfv --params-file /tmp/launch_params_k4fxv66q --params-file /tmp/launch_params_yhkbx9lp --params-file /tmp/launch_params_lj7xjp3n --params-file /tmp/launch_params_rqdn7507 --params-file /tmp/launch_params_h6h4z0sl'].
[INFO] [launch]: process[fsds_ros2_bridge-3] was required: shutting down launched system
[INFO] [fsds_ros2_bridge_camera-2]: sending signal 'SIGINT' to process[fsds_ros2_bridge_camera-2]
[INFO] [fsds_ros2_bridge_camera-1]: sending signal 'SIGINT' to process[fsds_ros2_bridge_camera-1]
[fsds_ros2_bridge_camera-2] [INFO] [1646947178.369877327] [rclcpp]: signal_handler(signal_value=2)
[fsds_ros2_bridge_camera-1] [INFO] [1646947178.371366329] [rclcpp]: signal_handler(signal_value=2)
[fsds_ros2_bridge_camera-1] Not all nodes were finished before finishing the context
[fsds_ros2_bridge_camera-1] .Ensure `rcl_node_fini` is called for all nodes before `rcl_context_fini`,to avoid leaking.
[fsds_ros2_bridge_camera-1] terminate called without an active exception
[fsds_ros2_bridge_camera-2] Not all nodes were finished before finishing the context
[fsds_ros2_bridge_camera-2] .Ensure `rcl_node_fini` is called for all nodes before `rcl_context_fini`,to avoid leaking.
[fsds_ros2_bridge_camera-2] terminate called without an active exception
[ERROR] [fsds_ros2_bridge_camera-1]: process has died [pid 68478, exit code -6, cmd '/home/user/Formula-Student-Driverless-Simulator/ros2/install/fsds_ros2_bridge/lib/fsds_ros2_bridge/fsds_ros2_bridge_camera --ros-args -r __node:=cam1 -r __ns:=/fsds/camera --params-file /tmp/launch_params_deymo1mh --params-file /tmp/launch_params_odo2kcxx --params-file /tmp/launch_params_vhostj66 --params-file /tmp/launch_params_o0imqskd'].
[ERROR] [fsds_ros2_bridge_camera-2]: process has died [pid 68480, exit code -6, cmd '/home/user/Formula-Student-Driverless-Simulator/ros2/install/fsds_ros2_bridge/lib/fsds_ros2_bridge/fsds_ros2_bridge_camera --ros-args -r __node:=cam2 -r __ns:=/fsds/camera --params-file /tmp/launch_params_dnt_rcdg --params-file /tmp/launch_params_oh2mfwq2 --params-file /tmp/launch_params_e1zg9fo9 --params-file /tmp/launch_params_mdcf1p1y'].

So basically the cameras are able to connect but the bridge is not.

My setup is the following:
Simulator is running on a desktop with the NVIDIA GPU
I am running the ros2_bridge on an Ubuntu virtual machine on my laptop.
To run the ros2_bridge I am using ROS_2 Galactic
I followed all the instructions on the [documentation](https://fs-driverless.github.io/Formula-Student-Driverless-Simulator/v2.1.0/getting-started-with-ros/) 

The ros_bridge (ROS1 Noetic) is working fine in the same setup.

Any ideas what is happening or what I might be missing?

Thanks for the help! :)

wouter-heerwegh, 2022-03-10:
Hi @GuihVicentini,

For the ROS 1 version, did you do any network setup? I know that in ROS1 I had some issues and I had to use the Host command to specify the ip of the machine running the simulator.

Have you tried this?

GuihVicentini, 2022-03-11:
Hi @wouter-heerwegh I might have expressed my self wrongly, the [ros1 bridge](https://github.com/FS-Driverless/Formula-Student-Driverless-Simulator/tree/master/ros) is working fine for me and yes, I pass the host argument to the command.

What is not working is the [ros2_bridge](https://github.com/FS-Driverless/Formula-Student-Driverless-Simulator/tree/master/ros2) and also here I pass the host argument in the cli:
**ros2 launch fsds_ros2_bridge fsds_ros2_bridge.launch.py host:=192.168.1.170**

As you can see from the logs I posted, the camera nodes are able to connect, so I don't think that the error on the network level, otherwise the cameras would not connect to the simulator.

First all 3 Nodes are waiting for the connection:
**[fsds_ros2_bridge-3] Waiting for connection -
[fsds_ros2_bridge_camera-2] Waiting for connection -
[fsds_ros2_bridge_camera-1] Waiting for connection -**

Than the bridge terminates but the cameras are able to connect:
**[fsds_ros2_bridge-3] terminate called after throwing an instance of 'std::runtime_error'
[fsds_ros2_bridge-3] what(): Failed connecting to rps client (airsim). Is the simulator running?
[fsds_ros2_bridge_camera-1] Connected to the simulator!
[fsds_ros2_bridge_camera-2] Connected to the simulator!**

wouter-heerwegh, 2022-03-11:
Could you verify that the rosbridge itself is correctly getting the host variable, I see that the bridge is using a different function to get the parameter. It should be localhost by default. If this is the case, you can replace it by the `declare_parameter` function in fsds_ros2_bridge_camera on line 152: `host_ip = nh->​declare_parameter​<std::string>(​"​host_ip​"​, ​"​localhost​"​);`

GuihVicentini, 2022-03-11:
Nice!! Yes, this did the trick, so it seems that the [ros2_bridge](https://github.com/FS-Driverless/Formula-Student-Driverless-Simulator/blob/master/ros2/src/fsds_ros2_bridge/src/fsds_ros2_bridge.cpp) is really not parsing the host parameter. 
I changed line 11 to
>host_ip = node->declare_parameter<std::string>("host_ip", "localhost");

Now things are working smoothly!

@wouter-heerwegh thank you so much for the fast responses!

closed 2022-03-11 by wouter-heerwegh