FS-Driverless/Formula-Student-Driverless-Simulator #247: Fatal Error when opening a map on Windows Release

MarvinKlemp, 2020-09-03:
I tryed out release `2.0.0` and `1.4.1` just getting a `fatal error` when launching a map.
Did I miss something during installation?

```
[2020.09.03-17.53.08:816][346]LogWindows: Error: === Critical error: ===
[2020.09.03-17.53.08:816][346]LogWindows: Error: 
[2020.09.03-17.53.08:816][346]LogWindows: Error: Fatal error!
[2020.09.03-17.53.08:816][346]LogWindows: Error: 
[2020.09.03-17.53.08:816][346]LogWindows: Error: Unhandled Exception: 0xe06d7363
[2020.09.03-17.53.08:816][346]LogWindows: Error: 
[2020.09.03-17.53.08:816][346]LogWindows: Error: [Callstack] 0x00007fff80383e49 KERNELBASE.dll!UnknownFunction []
[2020.09.03-17.53.08:816][346]LogWindows: Error: [Callstack] 0x00007fff6d286220 VCRUNTIME140.dll!UnknownFunction []
[2020.09.03-17.53.08:816][346]LogWindows: Error: [Callstack] 0x00007ff754ba4fc7 Blocks.exe!UnknownFunction []
[2020.09.03-17.53.08:816][346]LogWindows: Error: [Callstack] 0x00007ff754b96102 Blocks.exe!UnknownFunction []
[2020.09.03-17.53.08:816][346]LogWindows: Error: [Callstack] 0x00007ff7593168d1 Blocks.exe!UnknownFunction []
[2020.09.03-17.53.08:816][346]LogWindows: Error: [Callstack] 0x00007ff75922eb31 Blocks.exe!UnknownFunction []
[2020.09.03-17.53.08:816][346]LogWindows: Error: [Callstack] 0x00007ff7591dd99d Blocks.exe!UnknownFunction []
[2020.09.03-17.53.08:816][346]LogWindows: Error: [Callstack] 0x00007ff759252308 Blocks.exe!UnknownFunction []
[2020.09.03-17.53.08:816][346]LogWindows: Error: [Callstack] 0x00007ff7589cd860 Blocks.exe!UnknownFunction []
[2020.09.03-17.53.08:816][346]LogWindows: Error: [Callstack] 0x00007ff754ac5290 Blocks.exe!UnknownFunction []
[2020.09.03-17.53.08:816][346]LogWindows: Error: [Callstack] 0x00007ff754ad78fc Blocks.exe!UnknownFunction []
[2020.09.03-17.53.08:816][346]LogWindows: Error: [Callstack] 0x00007ff754ad79ba Blocks.exe!UnknownFunction []
[2020.09.03-17.53.08:816][346]LogWindows: Error: [Callstack] 0x00007ff754ae8c50 Blocks.exe!UnknownFunction []
[2020.09.03-17.53.08:816][346]LogWindows: Error: [Callstack] 0x00007ff75a283d0e Blocks.exe!UnknownFunction []
[2020.09.03-17.53.08:816][346]LogWindows: Error: [Callstack] 0x00007fff80a96fd4 KERNEL32.DLL!UnknownFunction []
[2020.09.03-17.53.08:816][346]LogWindows: Error: [Callstack] 0x00007fff8287cec1 ntdll.dll!UnknownFunction []
[2020.09.03-17.53.08:816][346]LogWindows: Error: 
[2020.09.03-17.53.08:836][346]LogExit: Executing StaticShutdownAfterError
[2020.09.03-17.53.08:838][346]LogWindows: FPlatformMisc::RequestExit(1)
[2020.09.03-17.53.08:838][346]LogCore: Engine exit requested (reason: Win RequestExit)
[2020.09.03-17.53.08:847][346]Log file closed, 09/03/20 19:53:08
```

davidoort, 2020-09-03:
Hi @MarvinKlemp ! Welcome to the project! 
So, on your windows computer you downloaded the release, extracted it, and executed Blocks.exe? Did you create a folder in your home directory called `Formula-Student-Driverless-Simulator` and pasted the `settings.json` file in this repository inside of it?

MarvinKlemp, 2020-09-04:
I'll close this one as this was an issue with the `settings.json`

Thanks for the fast reply :+1:

closed 2020-09-04 by MarvinKlemp