FS-Driverless/Formula-Student-Driverless-Simulator #248: Blank screen in windows

SaralTayal123, 2020-09-08:
Hey. I've tried everything but I'm still stuck with a black screen in windows with the pre-built binaries. 

I have made sure to have my settings.json file within the c/user/Formula-Student-Driverless-Simulator folder and I have also copied the release files into that folder and tried to run it from there. No matter which release I use, I run into the same issue. Any help would be appreciated.

Thanks

SijmenHuizenga, 2020-09-09:
Hi!

Can you try to run the simulator from a terminal with the `-log` flag like so:
```
.\FSDS.exe -log
```

With this you should be able to see what is going on in the terminal. Can you check for errors and post those logs here?

SaralTayal123, 2020-09-09:
LogInit: Build: ++UE4+Release-4.25-CL-13594126
LogInit: Engine Version: 4.25.1-13594126+++UE4+Release-4.25
LogInit: Compatible Engine Version: 4.25.0-13144385+++UE4+Release-4.25
LogInit: Net CL: 13144385
LogInit: OS: Windows 10 (Release 2004) (), CPU: Intel(R) Core(TM) i7-9750H CPU @ 2.60GHz, GPU: Intel(R) UHD Graphics 630
LogInit: Compiled (64-bit): Jun  2 2020 06:19:14
LogInit: Compiled with Visual C++: 19.16.27032.01
LogInit: Build Configuration: Development
LogInit: Branch Name: ++UE4+Release-4.25
LogInit: Command Line:  -log
LogInit: Base Directory: C:/Users/Saral/Formula-Student-Driverless-Simulator/FSOnline/Binaries/Win64/
LogInit: Allocator: binned2
LogInit: Installed Engine Build: 0
LogDevObjectVersion: Number of dev versions registered: 26
LogDevObjectVersion:   Dev-Blueprints (B0D832E4-1F89-4F0D-ACCF-7EB736FD4AA2): 10
LogDevObjectVersion:   Dev-Build (E1C64328-A22C-4D53-A36C-8E866417BD8C): 0
LogDevObjectVersion:   Dev-Core (375EC13C-06E4-48FB-B500-84F0262A717E): 4
LogDevObjectVersion:   Dev-Editor (E4B068ED-F494-42E9-A231-DA0B2E46BB41): 38
LogDevObjectVersion:   Dev-Framework (CFFC743F-43B0-4480-9391-14DF171D2073): 37
LogDevObjectVersion:   Dev-Mobile (B02B49B5-BB20-44E9-A304-32B752E40360): 2
LogDevObjectVersion:   Dev-Networking (A4E4105C-59A1-49B5-A7C5-40C4547EDFEE): 0
LogDevObjectVersion:   Dev-Online (39C831C9-5AE6-47DC-9A44-9C173E1C8E7C): 0
LogDevObjectVersion:   Dev-Physics (78F01B33-EBEA-4F98-B9B4-84EACCB95AA2): 4
LogDevObjectVersion:   Dev-Platform (6631380F-2D4D-43E0-8009-CF276956A95A): 0
LogDevObjectVersion:   Dev-Rendering (12F88B9F-8875-4AFC-A67C-D90C383ABD29): 43
LogDevObjectVersion:   Dev-Sequencer (7B5AE74C-D270-4C10-A958-57980B212A5A): 12
LogDevObjectVersion:   Dev-VR (D7296918-1DD6-4BDD-9DE2-64A83CC13884): 3
LogDevObjectVersion:   Dev-LoadTimes (C2A15278-BFE7-4AFE-6C17-90FF531DF755): 1
LogDevObjectVersion:   Private-Geometry (6EACA3D4-40EC-4CC1-B786-8BED09428FC5): 3
LogDevObjectVersion:   Dev-AnimPhys (29E575DD-E0A3-4627-9D10-D276232CDCEA): 17
LogDevObjectVersion:   Dev-Anim (AF43A65D-7FD3-4947-9873-3E8ED9C1BB05): 7
LogDevObjectVersion:   Dev-ReflectionCapture (6B266CEC-1EC7-4B8F-A30B-E4D90942FC07): 1
LogDevObjectVersion:   Dev-Automation (0DF73D61-A23F-47EA-B727-89E90C41499A): 1
LogDevObjectVersion:   FortniteMain (601D1886-AC64-4F84-AA16-D3DE0DEAC7D6): 31
LogDevObjectVersion:   Dev-Enterprise (9DFFBCD6-494F-0158-E221-12823C92A888): 10
LogDevObjectVersion:   Dev-Niagara (F2AED0AC-9AFE-416F-8664-AA7FFA26D6FC): 1
LogDevObjectVersion:   Dev-Destruction (174F1F0B-B4C6-45A5-B13F-2EE8D0FB917D): 10
LogDevObjectVersion:   Dev-Physics-Ext (35F94A83-E258-406C-A318-09F59610247C): 37
LogDevObjectVersion:   Dev-PhysicsMaterial-Chaos (B68FC16E-8B1B-42E2-B453-215C058844FE): 1
LogDevObjectVersion:   Dev-CineCamera (B2E18506-4273-CFC2-A54E-F4BB758BBA07): 1
LogInit: Presizing for max 2097152 objects, including 1 objects not considered by GC, pre-allocating 0 bytes for permanent pool.
LogConfig: Applying CVar settings from Section [/Script/Engine.StreamingSettings] File [../../../FSOnline/Saved/Config/WindowsNoEditor/Engine.ini]
LogConfig: Setting CVar [[s.MinBulkDataSizeForAsyncLoading:131072]]
LogConfig: Setting CVar [[s.AsyncLoadingThreadEnabled:0]]
LogConfig: Setting CVar [[s.EventDrivenLoaderEnabled:1]]
LogConfig: Setting CVar [[s.WarnIfTimeLimitExceeded:0]]
LogConfig: Setting CVar [[s.TimeLimitExceededMultiplier:1.5]]
LogConfig: Setting CVar [[s.TimeLimitExceededMinTime:0.005]]
LogConfig: Setting CVar [[s.UseBackgroundLevelStreaming:1]]
LogConfig: Setting CVar [[s.PriorityAsyncLoadingExtraTime:15.0]]
LogConfig: Setting CVar [[s.LevelStreamingActorsUpdateTimeLimit:5.0]]
LogConfig: Setting CVar [[s.PriorityLevelStreamingActorsUpdateExtraTime:5.0]]
LogConfig: Setting CVar [[s.LevelStreamingComponentsRegistrationGranularity:10]]
LogConfig: Setting CVar [[s.UnregisterComponentsTimeLimit:1.0]]
LogConfig: Setting CVar [[s.LevelStreamingComponentsUnregistrationGranularity:5]]
LogConfig: Setting CVar [[s.FlushStreamingOnExit:1]]
LogStreaming: Display: Async Loading initialized: Event Driven Loader: true, Async Loading Thread: false, Async Post Load: true
LogInit: Object subsystem initialized
LogConfig: Setting CVar [[con.DebugEarlyDefault:1]]
LogConfig: Setting CVar [[r.setres:1280x720]]
LogConfig: Setting CVar [[fx.NiagaraAllowRuntimeScalabilityChanges:1]]
[2020.09.09-22.54.29:102][  0]LogConfig: Setting CVar [[con.DebugEarlyDefault:1]]
[2020.09.09-22.54.29:102][  0]LogConfig: Setting CVar [[r.setres:1280x720]]
[2020.09.09-22.54.29:103][  0]LogConfig: Setting CVar [[fx.NiagaraAllowRuntimeScalabilityChanges:1]]
[2020.09.09-22.54.29:103][  0]LogConfig: Applying CVar settings from Section [/Script/Engine.RendererSettings] File [../../../FSOnline/Saved/Config/WindowsNoEditor/Engine.ini]
[2020.09.09-22.54.29:103][  0]LogConfig: Setting CVar [[r.GPUCrashDebugging:0]]
[2020.09.09-22.54.29:104][  0]LogConfig: Applying CVar settings from Section [/Script/Engine.RendererOverrideSettings] File [../../../FSOnline/Saved/Config/WindowsNoEditor/Engine.ini]
[2020.09.09-22.54.29:104][  0]LogConfig: Applying CVar settings from Section [/Script/Engine.StreamingSettings] File [../../../FSOnline/Saved/Config/WindowsNoEditor/Engine.ini]
[2020.09.09-22.54.29:105][  0]LogConfig: Setting CVar [[s.MinBulkDataSizeForAsyncLoading:131072]]
[2020.09.09-22.54.29:105][  0]LogConfig: Setting CVar [[s.AsyncLoadingThreadEnabled:0]]
[... 594 lines cut ...]
[... cut ...]
[2020.09.09-22.54.32:189][  0]LogSlate: InvalidateAllWidgets triggered.  All widgets were invalidated
[2020.09.09-22.54.32:191][  0]LogSlate: Slate font cache was flushed
[2020.09.09-22.54.32:195][  1]LogSlate: Took 0.000964 seconds to synchronously load lazily loaded font '../../../Engine/Content/Slate/Fonts/Roboto-Regular.ttf' (155K)
[2020.09.09-22.54.32:196][  1]LogSlate: Took 0.000908 seconds to synchronously load lazily loaded font '../../../Engine/Content/EngineFonts/Faces/RobotoBold.ufont' (160K)

SaralTayal123, 2020-09-09:
Sorry. This is a very long log file.

This line towards the end was highlighted in red as an error `[2020.09.09-22.54.32:083][  0]LogPlayerController: Error: InputMode:UIOnly - Attempting to focus Non-Focusable widget SObjectWidget [Widget.cpp(799)]!`

Hope this help. Thanks

SijmenHuizenga, 2020-09-10:
Thanks! The error `Attempting to focus Non-Focusable widget SObjectWidget` does also show on my computer but here it does work. The rest of the logs also look fine. I can't reproduce the problem, without this, it is impossible to debug. 

A shot in the dark: I changed a bit of in the UI configuration. [Download here](https://drive.google.com/file/d/1_mhe7GAdGuZoI0VToNhFubpbLGuwtatW/view?usp=sharing). Can you try this version and see if it improves? If you get a black screen, can you try to press Enter as soon as the simulation starts? See if that does anything?

If that doesn't work, can you try to run `FSDS.exe /Game/TrainingMap`? This should skip the main menu and launch the simulation at once.

SaralTayal123, 2020-09-10:
Thanks @SijmenHuizenga I'm unable to download the updated UI configuration. Google drive tells me I have insufficient permissions. I have requested for permissions from google drive so I think you should get an email about that.

I did, however, try the pressing 'enter' on the black screen without luck on the old version of the UI. Will try that with the new version once I am able to download it. 

The `FSDS.exe /Game/TrainingMap` doesn't work either :( (It launches the window but its still black). 

One interesting think I noticed is that the cursor changes to one of a text editing cursor towards the upper-center of the screen. This and the high gpu/cpu utilization makes me believe that the simulation is opening in the background but the black screen is just open on top. Not sure. 

BTW, This is running on a relatively fresh install of windows. Do you think I am missing any dependencies?

SijmenHuizenga, 2020-09-10:
Sorry about those permissions, you should be able to access the file now.

Can you check the output of `vulkaninfo` in terminal to see if your graphics drivers are installed correctly?

SaralTayal123, 2020-09-10:
Sadly still no luck. I have the vulcan drivers and the windows equivalent of `vulcaninfo` works and correctly reports the GPU drivers. 

The log files also seem to indicate that the track is correctly launched via FSDS.exe /Game/TrainingMap.

Sifting through the log files I noticed that this line at the end `[2020.09.10-14.29.20:222][  0]LogSlate: InvalidateAllWidgets triggered.  All widgets were invalidated`

Do you think this could be the cause of nothing showing up on the screen?

SijmenHuizenga, 2020-09-11:
Thats unfortunate :( Super weird that the track loads but nothing is shown. I'm a bit out of ideas.

These are my logs, on windows. As you can see, the InvalidateAllWidgets log line also exists here.
<details>
<summary>show logs</summary>
LogInit: Build: ++UE4+Release-4.25-CL-13594126
LogInit: Engine Version: 4.25.1-13594126+++UE4+Release-4.25
LogInit: Compatible Engine Version: 4.25.0-13144385+++UE4+Release-4.25
LogInit: Net CL: 13144385
LogInit: OS: Windows 10 (Release 2004) (), CPU: AMD Ryzen 7 3700X 8-Core Processor             , GPU: NVIDIA GeForce GTX 1660
LogInit: Compiled (64-bit): Jun  2 2020 06:19:14
LogInit: Compiled with Visual C++: 19.16.27032.01
LogInit: Build Configuration: Development
LogInit: Branch Name: ++UE4+Release-4.25
LogInit: Command Line:  -log
LogInit: Base Directory: C:/Users/sijme/Desktop/exports/2.0.0/fsds-v2.0.0-win/FSOnline/Binaries/Win64/
LogInit: Allocator: binned2
LogInit: Installed Engine Build: 0
LogDevObjectVersion: Number of dev versions registered: 26
LogDevObjectVersion:   Dev-Blueprints (B0D832E4-1F89-4F0D-ACCF-7EB736FD4AA2): 10
LogDevObjectVersion:   Dev-Build (E1C64328-A22C-4D53-A36C-8E866417BD8C): 0
LogDevObjectVersion:   Dev-Core (375EC13C-06E4-48FB-B500-84F0262A717E): 4
LogDevObjectVersion:   Dev-Editor (E4B068ED-F494-42E9-A231-DA0B2E46BB41): 38
LogDevObjectVersion:   Dev-Framework (CFFC743F-43B0-4480-9391-14DF171D2073): 37
LogDevObjectVersion:   Dev-Mobile (B02B49B5-BB20-44E9-A304-32B752E40360): 2
LogDevObjectVersion:   Dev-Networking (A4E4105C-59A1-49B5-A7C5-40C4547EDFEE): 0
LogDevObjectVersion:   Dev-Online (39C831C9-5AE6-47DC-9A44-9C173E1C8E7C): 0
LogDevObjectVersion:   Dev-Physics (78F01B33-EBEA-4F98-B9B4-84EACCB95AA2): 4
LogDevObjectVersion:   Dev-Platform (6631380F-2D4D-43E0-8009-CF276956A95A): 0
LogDevObjectVersion:   Dev-Rendering (12F88B9F-8875-4AFC-A67C-D90C383ABD29): 43
LogDevObjectVersion:   Dev-Sequencer (7B5AE74C-D270-4C10-A958-57980B212A5A): 12
LogDevObjectVersion:   Dev-VR (D7296918-1DD6-4BDD-9DE2-64A83CC13884): 3
LogDevObjectVersion:   Dev-LoadTimes (C2A15278-BFE7-4AFE-6C17-90FF531DF755): 1
LogDevObjectVersion:   Private-Geometry (6EACA3D4-40EC-4CC1-B786-8BED09428FC5): 3
LogDevObjectVersion:   Dev-AnimPhys (29E575DD-E0A3-4627-9D10-D276232CDCEA): 17
LogDevObjectVersion:   Dev-Anim (AF43A65D-7FD3-4947-9873-3E8ED9C1BB05): 7
LogDevObjectVersion:   Dev-ReflectionCapture (6B266CEC-1EC7-4B8F-A30B-E4D90942FC07): 1
LogDevObjectVersion:   Dev-Automation (0DF73D61-A23F-47EA-B727-89E90C41499A): 1
LogDevObjectVersion:   FortniteMain (601D1886-AC64-4F84-AA16-D3DE0DEAC7D6): 31
LogDevObjectVersion:   Dev-Enterprise (9DFFBCD6-494F-0158-E221-12823C92A888): 10
LogDevObjectVersion:   Dev-Niagara (F2AED0AC-9AFE-416F-8664-AA7FFA26D6FC): 1
LogDevObjectVersion:   Dev-Destruction (174F1F0B-B4C6-45A5-B13F-2EE8D0FB917D): 10
LogDevObjectVersion:   Dev-Physics-Ext (35F94A83-E258-406C-A318-09F59610247C): 37
LogDevObjectVersion:   Dev-PhysicsMaterial-Chaos (B68FC16E-8B1B-42E2-B453-215C058844FE): 1
LogDevObjectVersion:   Dev-CineCamera (B2E18506-4273-CFC2-A54E-F4BB758BBA07): 1
LogInit: Presizing for max 2097152 objects, including 1 objects not considered by GC, pre-allocating 0 bytes for permanent pool.
LogConfig: Applying CVar settings from Section [/Script/Engine.StreamingSettings] File [../../../FSOnline/Saved/Config/WindowsNoEditor/Engine.ini]
LogConfig: Setting CVar [[s.MinBulkDataSizeForAsyncLoading:131072]]
LogConfig: Setting CVar [[s.AsyncLoadingThreadEnabled:0]]
LogConfig: Setting CVar [[s.EventDrivenLoaderEnabled:1]]
LogConfig: Setting CVar [[s.WarnIfTimeLimitExceeded:0]]
LogConfig: Setting CVar [[s.TimeLimitExceededMultiplier:1.5]]
LogConfig: Setting CVar [[s.TimeLimitExceededMinTime:0.005]]
LogConfig: Setting CVar [[s.UseBackgroundLevelStreaming:1]]
LogConfig: Setting CVar [[s.PriorityAsyncLoadingExtraTime:15.0]]
LogConfig: Setting CVar [[s.LevelStreamingActorsUpdateTimeLimit:5.0]]
LogConfig: Setting CVar [[s.PriorityLevelStreamingActorsUpdateExtraTime:5.0]]
LogConfig: Setting CVar [[s.LevelStreamingComponentsRegistrationGranularity:10]]
LogConfig: Setting CVar [[s.UnregisterComponentsTimeLimit:1.0]]
LogConfig: Setting CVar [[s.LevelStreamingComponentsUnregistrationGranularity:5]]
LogConfig: Setting CVar [[s.FlushStreamingOnExit:1]]
LogStreaming: Display: Async Loading initialized: Event Driven Loader: true, Async Loading Thread: false, Async Post Load: true
LogInit: Object subsystem initialized
LogConfig: Setting CVar [[con.DebugEarlyDefault:1]]
LogConfig: Setting CVar [[r.setres:1280x720]]
LogConfig: Setting CVar [[fx.NiagaraAllowRuntimeScalabilityChanges:1]]
[2020.09.11-12.02.51:520][  0]LogConfig: Setting CVar [[con.DebugEarlyDefault:1]]
[2020.09.11-12.02.51:521][  0]LogConfig: Setting CVar [[r.setres:1280x720]]
[2020.09.11-12.02.51:521][  0]LogConfig: Setting CVar [[fx.NiagaraAllowRuntimeScalabilityChanges:1]]
[2020.09.11-12.02.51:521][  0]LogConfig: Applying CVar settings from Section [/Script/Engine.RendererSettings] File [../../../FSOnline/Saved/Config/WindowsNoEditor/Engine.ini]
[... 531 lines cut ...]
[2020.09.11-12.02.52:630][  0]LogWorld: Bringing up level for play took: 0.000202
[2020.09.11-12.02.52:631][  0]LogGameMode: FindPlayerStart: PATHS NOT DEFINED or NO PLAYERSTART with positive rating
[2020.09.11-12.02.52:631][  0]LogGameMode: Display: Match State Changed from EnteringMap to WaitingToStart
[2020.09.11-12.02.52:631][  0]LogGameState: Match State Changed from EnteringMap to WaitingToStart
[... cut ...]
[2020.09.11-12.02.52:737][  0]LogSlate: Slate font cache was flushed
[2020.09.11-12.02.52:740][  1]LogSlate: Took 0.000651 seconds to synchronously load lazily loaded font '../../../Engine/Content/Slate/Fonts/Roboto-Regular.ttf' (155K)
[2020.09.11-12.02.52:741][  1]LogSlate: Took 0.000633 seconds to synchronously load lazily loaded font '../../../Engine/Content/EngineFonts/Faces/RobotoBold.ufont' (160K)
</details>

Sorry, I messed up spelling. I wanted to ask if `vulkaninfo` outputs correctly. Can you check if vulkan is installed correctly?

Nirav-Mange, 2020-10-16:
is this bug solved? I had the same issue and my laptop kept crashing

SijmenHuizenga, 2020-10-16:
Hi @Nirav-Mange, unfortunately we haven't been able to solve this issue. It seems like nobody knows what causes it. Help & ideas are very much welcome!

Liam-Ferrante, 2021-03-30:
When I had the black screen with no buttons it was because FSDS was automatically running in VR. Putting on my Oculus headset was necessary to see the simulation. You can change this automatic VR setting in Unreal though.

AlexFigas, 2024-04-28:
Same issue here.

wouter-heerwegh, 2024-04-28:
Hi @AlexFigas, 

Could you provide some logs from the executable and some more information about your system? There was a similar issue in the past, but that was related to the settings.json file.

AlexFigas, 2024-04-30:
> Hi @AlexFigas,
> 
> Could you provide some logs from the executable and some more information about your system? There was a similar issue in the past, but that was related to the settings.json file.

I've fixed the issue by running the simulator on windows directly instead of inside wsl.
Probably will give me some issues down the road since I see that it doesn't work that well and has some issues with windows.
But thanks @wouter-heerwegh

wouter-heerwegh, 2024-04-30:
Hi @AlexFigas,

Good to hear, if you're running on windows, I'd indeed recommend running the executable in windows. But you can still run the bridge in WSL.

If you're using WSL2, you just have to run the rosbridge with the host parameter set to the ip on your windows machine.