FS-Driverless/Formula-Student-Driverless-Simulator #256: Testing only topics reliability

manderino1, 2020-11-25:
I'm trying to use the fsds/testing_only/cones and odom topics to test the precision of my calculated cone poses but, when I plot them in rviz, I find them shifted like map was shifted down with respect to the odometry.
Is this a problem into the sim or something wrong by me while plotting them? Thank you!
(For reference, to get a correct map i have in the training map to shift down x axis by like 0.8 meters, I know that they are with respect to the starting position, but the starting position in /fsds/testing_only/odom is about 0 on the x axis)

[image]
[image]

SijmenHuizenga, 2020-11-25:
@davidoort any idea what's going on here?

davidoort, 2020-11-28:
This is the famous x-offset issue that we struggled with for a while (the one that made the car drive through cones when it thought it was following the midline). I seem to remember that this was fixed at the competition, so not sure why this is popping up again. @manderino1 could you test this without any changes in the ros wrapper for different maps and see if you get the same issue?

manderino1, 2020-11-29:
@davidoort I tried also other mapsbut it seems to be on every one shifted about 2 metres on the x-axis
If I edit my plotting node with (x = x-2) it's almost perfect, but if I simply plot the topic it's clearly shifted.

davidoort, 2020-11-30:
Okay, we can always include this offset in the wrapper as a quick fix, but it is not very elegant. @SijmenHuizenga I believe this was something related to the cg definition of the car in Unreal. If we could fix this offset in the blueprint that loads this initial position I think it would be better

wouter-heerwegh, 2020-12-22:
Does this onoy happen in a specifc case or version? I've been using the testing topics for a couple of months and have not seen this before.

ckurtz22, 2021-01-14:
I would expect that this is some sort of reference frame issue.

ckurtz22, 2021-01-14:
> This is the famous x-offset issue that we struggled with for a while (the one that made the car drive through cones when it thought it was following the midline). I seem to remember that this was [...]

Looks like the relevant issue being referred to here is #144

MarvinKlemp, 2021-02-17:
I can confirm that I also encountered this issue. Subtracted x-2 for all initial cone positions fix this.

closed 2022-03-09 by wouter-heerwegh