FS-Driverless/Formula-Student-Driverless-Simulator #257: Topic with extra sim info

wouter-heerwegh, 2020-12-22:
Would it be possible to have information from the simulator, like for instance the amount of cones hit?

Not sure about what other info you could pass through, but I'm sure someone else can find other info they would like.

SijmenHuizenga, 2020-12-22:
There is an [undocumented feature](https://github.com/FS-Driverless/Formula-Student-Driverless-Simulator/blob/master/python/fsds/client.py#L197) in the python api called `.getRefereeState()`. It returns
* `doo_counter` = the number of cones down or out. In unreal this is implemented as 'number of cones touched by the car', so it is a bit different from the FSG DOO definition. 
* `lap_times` = the number of finished laps.

These methods [are used](https://github.com/FS-Driverless/Formula-Student-Driverless-Simulator/blob/master/operator/operator.py#L105) by the operator (used during FSOnline) to calculate scores.

SijmenHuizenga, 2021-01-17:
This was added in #260 and merged in #261. Thanks so much for implementing this @wouter-heerwegh!

closed 2021-01-17 by SijmenHuizenga