FS-Driverless/Formula-Student-Driverless-Simulator #258: Move ROS workspace to its own repository

ckurtz22, 2020-12-24:
Would be nice to be able to just clone the ros bridge part into someone's current workspace. It seems to me that the only issues is the AirSim dependencies, so these would have to be adjusted to be in the ros workspace instead of the repository for the unreal project.

SijmenHuizenga, 2021-01-17:
The dependencies on the AirSim library (aka AirLib) are quite significant. Data structure definitions are shared between the simulator and the ros bridge. Doing cross-repository dependency management can get quite painful when updates need to be applied to both to implement a new feature. 

I do see that cloning the whole repository to just use the ros bridge is not that convenient. So if we can find a way to fix this without introducing dependency problems I would be all for it.

mateusz-lichota, 2022-07-21:
How about making a special "2.1.0_ros2_bridge_only" branch every release with everything not needed by the bridge deleted? That way people could only get the bridge without downloading hundreds of megabytes of additional data. (Using `git clone -b mybranch --single-branch git://sub.domain.com/repo.git`