FS-Driverless/Formula-Student-Driverless-Simulator #262: UE Project not opening on Ubuntu

FirenzeRaceTeam, 2021-02-08:
I followed all the instructions for building from source, clicking "Skip conversion" and then "Yes" to rebuild AirSim and Blocks, but still I get the "FSOnline could not be compiled" error.

SijmenHuizenga, 2021-02-08:
Is there an error message in the console?

FirenzeRaceTeam, 2021-02-08:
It wouldn't compile with UE 4.26 because it used deprecated code, but on 4.25 I was able to get it to compile, but it still won't start. Here's what the console says:
```
[2021.02.08-19.07.10:221][  0]LogLinker: Warning: The file '../../../../Formula-Student-Driverless-Simulator/UE4Project/Plugins/AirSim/Content/Blueprints/AirsimSpectatorPawn.uasset' contains unrecognizable data, check that it is of the expected type.
[2021.02.08-19.07.10:221][  0]LogLinker: Warning: The file '../../../../Formula-Student-Driverless-Simulator/UE4Project/Plugins/AirSim/Content/Blueprints/AirsimSpectatorPawn.uasset' contains unrecognizable data, check that it is of the expected type.
[2021.02.08-19.07.10:221][  0]LogUObjectGlobals: Warning: Failed to find object 'Class /AirSim/Blueprints/AirsimSpectatorPawn.AirsimSpectatorPawn_C'
CDO Constructor (AirSimGameMode): Failed to find /AirSim/Blueprints/AirsimSpectatorPawn.AirsimSpectatorPawn_C
[2021.02.08-19.07.10:221][  0]Error: CDO Constructor (AirSimGameMode): Failed to find /AirSim/Blueprints/AirsimSpectatorPawn.AirsimSpectatorPawn_C

[2021.02.08-19.07.10:221][  0]Error: CDO Constructor (AirSimGameMode): Failed to find /AirSim/Blueprints/AirsimSpectatorPawn.AirsimSpectatorPawn_C

[2021.02.08-19.07.10:221][  0]LogLinker: Warning: The file '../../../../Formula-Student-Driverless-Simulator/UE4Project/Plugins/AirSim/Content/Blueprints/BP_PIPCamera.uasset' contains unrecognizable data, check that it is of the expected type.
[2021.02.08-19.07.10:221][  0]LogLinker: Warning: The file '../../../../Formula-Student-Driverless-Simulator/UE4Project/Plugins/AirSim/Content/Blueprints/BP_PIPCamera.uasset' contains unrecognizable data, check that it is of the expected type.
[2021.02.08-19.07.10:221][  0]LogUObjectGlobals: Warning: Failed to find object 'Class /AirSim/Blueprints/BP_PIPCamera.BP_PIPCamera_C'
CDO Constructor (CarPawn): Failed to find /AirSim/Blueprints/BP_PIPCamera.BP_PIPCamera_C
[2021.02.08-19.07.10:221][  0]Error: CDO Constructor (CarPawn): Failed to find /AirSim/Blueprints/BP_PIPCamera.BP_PIPCamera_C

[2021.02.08-19.07.10:221][  0]Error: CDO Constructor (CarPawn): Failed to find /AirSim/Blueprints/BP_PIPCamera.BP_PIPCamera_C

[2021.02.08-19.07.10:221][  0]LogLinker: Warning: The file '../../../../Formula-Student-Driverless-Simulator/UE4Project/Plugins/AirSim/Content/VehicleAdv/WheelData/Slippery.uasset' contains unrecognizable data, check that it is of the expected type.
[2021.02.08-19.07.10:221][  0]LogLinker: Warning: The file '../../../../Formula-Student-Driverless-Simulator/UE4Project/Plugins/AirSim/Content/VehicleAdv/WheelData/Slippery.uasset' contains unrecognizable data, check that it is of the expected type.
[2021.02.08-19.07.10:221][  0]LogUObjectGlobals: Warning: Failed to find object 'Object /AirSim/VehicleAdv/WheelData/Slippery.Slippery'
terminating with uncaught exception of type std::invalid_argument: Failed to load asset object - /AirSim/VehicleAdv/WheelData/Slippery.Slippery
Signal 6 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=6
[2021.02.08-19.07.10:240][  0]LogCore: === Critical error: ===
Unhandled Exception: SIGABRT: abort() called

[2021.02.08-19.07.10:240][  0]LogCore: Fatal error!

0x00007fbabd24818b libc.so.6!gsignal(+0xcb)
0x00007fbabd227859 libc.so.6!abort(+0x12a)
0x00007fbac14b107b libUE4Editor-InputCore.so!abort_message()
0x00007fbac149726e libUE4Editor-InputCore.so!demangling_terminate_handler()()
0x00007fbac1496fb3 libUE4Editor-InputCore.so!std::__terminate(void (*)())()
0x00007fbac14b2de6 libUE4Editor-InputCore.so!__cxxabiv1::failed_throw(__cxxabiv1::__cxa_exception*)()
0x00007fbac14b2d7f libUE4Editor-InputCore.so!__cxa_throw(+0x6e)
0x00007fba36036bf1 libUE4Editor-AirSim.so!UAirBlueprintLib::LoadObject(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) [/home/frt-driverless/Formula-Student-Driverless-Simulator/UE4Project/Plugins/AirSim/Source/AirBlueprintLib.cpp:604]
0x00007fba3606a0c3 libUE4Editor-AirSim.so!ACarPawn::ACarPawn() [/home/frt-driverless/Formula-Student-Driverless-Simulator/UE4Project/Plugins/AirSim/Source/Vehicles/Car/CarPawn.cpp:25]
0x00007fbac5be662d libUE4Editor-CoreUObject.so!UClass::CreateDefaultObject() [/home/frt-driverless/UnrealEngine/Engine/Source/Runtime/CoreUObject/Private/UObject/Class.cpp:3672]
0x00007fbac5e3dc88 libUE4Editor-CoreUObject.so!ProcessNewlyLoadedUObjects(FName, bool) [/home/frt-driverless/UnrealEngine/Engine/Source/Runtime/CoreUObject/Private/UObject/UObjectBase.cpp:949]
0x00007fbac5d8862e libUE4Editor-CoreUObject.so!TBaseStaticDelegateInstance<void (FName, bool)>::ExecuteIfSafe(FName, bool) const [/home/frt-driverless/UnrealEngine/Engine/Source/Runtime/Core/Public/Delegates/DelegateInstancesImpl.h:857]
0x00007fbac6778d43 libUE4Editor-Core.so!TBaseMulticastDelegate<void, FName, bool>::Broadcast(FName, bool) const [/home/frt-driverless/UnrealEngine/Engine/Source/Runtime/Core/Public/Delegates/DelegateSignatureImpl.inl:1013]
0x00007fbac674a3c4 libUE4Editor-Core.so!FModuleManager::LoadModuleWithFailureReason(FName, EModuleLoadResult&) [/home/frt-driverless/UnrealEngine/Engine/Source/Runtime/Core/Private/Modules/ModuleManager.cpp:497]
0x00007fbac1397e03 libUE4Editor-Projects.so!FModuleDescriptor::LoadModulesForPhase(ELoadingPhase::Type, TArray<FModuleDescriptor, TSizedDefaultAllocator<32> > const&, TMap<FName, EModuleLoadResult, FDefaultSetAllocator, TDefaultMapHashableKeyFuncs<FName, EModuleLoadResult, false> >&) [/hom [... cut ...]
0x00000000002805ec UE4Editor!FEngineLoop::PreInitPostStartupScreen(char16_t const*) [/home/frt-driverless/UnrealEngine/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:3220]
0x000000000026f31b UE4Editor!GuardedMain(char16_t const*) [/home/frt-driverless/UnrealEngine/Engine/Source/Runtime/Launch/Private/Launch.cpp:127]
0x00007fbabd402b5b libUE4Editor-UnixCommonStartup.so!CommonUnixMain(int, char**, int (*)(char16_t const*), void (*)()) [/home/frt-driverless/UnrealEngine/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:264]
0x00007fbabd2290b3 libc.so.6!__libc_start_main(+0xf2)
0x000000000025d029 UE4Editor!_start()
```

SijmenHuizenga, 2021-02-08:
Can you check the content of `Formula-Student-Driverless-Simulator/UE4Project/Plugins/AirSim/Content/Blueprints/AirsimSpectatorPawn.uasset`?

FirenzeRaceTeam, 2021-02-08:
```
version https://git-lfs.github.com/spec/v1
oid sha256:a5933b422e11008ff81d6e0730204a6f7dd531c19887135372b5bbe626e1a001
size 164381
```

SijmenHuizenga, 2021-02-08:
Ah I see, you did not have git lfs installed when cloning. Run `git lfs fetch` and `git lfs pull` to download and install the files.

FirenzeRaceTeam, 2021-02-10:
Thank you, I forgot to pull the files.

closed 2021-02-10 by FirenzeRaceTeam