FS-Driverless/Formula-Student-Driverless-Simulator #265: Calibration info for cameras

niktnikt, 2021-02-25:
Hi,
Is it possible to provide camera calibration info to the simulator similar to what is here (http://docs.ros.org/en/melodic/api/sensor_msgs/html/msg/CameraInfo.html)? If yes, is it also possible to extract it?

floriankittelmann, 2022-04-28:
Hi all,

This might be an old issue, but I am also quite interested in having a CameraInfo Topic for depth camera. I need it because the current depth camera just outputs a Depth Image, but for my application I need a PointCloud with x,y,z coordinates. As I see from he documentation, there is already an existing package in ROS which is doing that (https://wiki.ros.org/depth_image_proc#depth_image_proc.2Fpoint_cloud_xyz), but it needs the CameraInfo Topic to get the correct x,y,z coordinates. 

However, I am thinking about to add the CameraInfo Topic and make a PR. But I am not sure where or how I retrieve the correct information for the CameraInfo Topic. What are the informations for Intrinsic camera matrix (K), Rectification matrix (R), Projection/camera matrix (P), etc? These information seems for me to be sensor/camera specific and therefore I will need know how the camera/sensor gets simulated in the simulator..

Thanks in advance

wouter-heerwegh, 2022-05-04:
Hi @floriankittelmann,

I don't have a clear idea of how to get this info yet, but I saw that this is implemented in [Airsim] (https://github.com/microsoft/AirSim/blob/master/ros/src/airsim_ros_pkgs/src/airsim_ros_wrapper.cpp) already. This might give some idea of how to get this info. It could be that our version does not support this yet as we are on Airsim 1.2 (I think).