FS-Driverless/Formula-Student-Driverless-Simulator #269: Not precise depth values

federicochiesa, 2021-06-07:
We have enabled the depth camera on the simulator, but the depth values are given as integer and are not precise. For example, in [this](https://imgur.com/a/dEkQgQA) image, the yellow cone on the right is given a depth of "5", the blue one is "6" and the yellow one in the distance is "15". This seems wrong as the cones are very close to the car's wheel and it's strange to see a depth of 5(meters?) from the camera.

The RGB image that I sent corresponds to the depth one, we set the two cameras to be in the same position in the settings.json

mvanlobensels, 2021-08-04:
Hi @federicochiesa, from what I can remember, these depth sensors were implemented in a short timeframe very close to the start date of the FS Online competition. They also were pretty overpowered compared to teams not using them, so it could be that they were nerfed by making them less accurate. @SijmenHuizenga Please correct me if I am wrong.

mvanlobensels, 2021-09-10:
I'm don't think there's a flag for it. I'm afraid you'll have to dive into the sensor code yourself to fix it

masskro0, 2022-11-28:
I'm using the depth cameras as well and they provide me accurate values.
Could you share how you calculate the depth of the cones? Maybe there is a bug in your implementation.