FS-Driverless/Formula-Student-Driverless-Simulator #272: Ubuntu 20.04

ydmunck, 2021-09-01:
[image]

I managed to open the Simulator on Ubuntu 20.04 in Unreal Engine 4.25. However when I start the simulator by pressing "Play", the simulator crashes with the error message above. I would like to run the simulator in Ubuntu 20.04 to investigate possible methods to transfer the ROS bridge to ROS noetic. Would this be possible?

wouter-heerwegh, 2021-09-09:
Hi @ydmunck,

I never got unreal running properly on linux, but what I did was run it on windows and running the bridge in wsl (keep in mind, not wsl2, thats a big mess). 

Don't know if you got any further by now but, from the log it looks like it got an error while reading a settings file. Could be a wrong type for instance. Try to keep your settings as minimalistic as possible for now, and try to modify it step by step.

Hope you can figure it out!

wouter-heerwegh, 2021-09-16:
@ydmunck  I was able to build and run the ros_bridge in Noetic. The only thing I had to do was change the first line of the camera_launcher.py file to point to python3 instead of just python

closed 2022-09-03 by mateusz-lichota