FS-Driverless/Formula-Student-Driverless-Simulator #277: Blocks and AirSim cannot be built (Ubuntu 18.04 + UE4.27.1)

KnockerPulsar, 2021-11-06:
Hello, thanks for the great work.
I've been trying to get the simulator working on Ubuntu for a while now. The binaries work perfectly. But since I might need to play around with some things, I need to work with the UE project.

I've followed [these](https://docs.unrealengine.com/4.27/en-US/SharingAndReleasing/Linux/BeginnerLinuxDeveloper/SettingUpAnUnrealWorkflow/) instructions to get UE4 running on my machine.
Following the the instructions in [getting started](https://github.com/FS-Driverless/Formula-Student-Driverless-Simulator/blob/master/docs/getting-started.md), I downloaded the source code from the latest release, everything goes well until I have to start the project. When prompted to convert the project, I skip. Then when prompted to build Airsim and Blocks, I choose "yes". But the engine fails, telling me that it cannot build them while the engine is running and to rebuild from the IDE.
When I check the terminal output, I can see and error saying that building would modify engine files, then a long list of UE4 engine files.
I tried converting the project, but that fails as well.

My main suspicion is that the version I'm using is not compatible. Can you kindly inform me which versions you tested on?

wouter-heerwegh, 2021-11-06:
Hi @KnockerPulsar,

I'm not exactly sure which exact version was used, I'm still trying to compile new binaries for Linux, but I can open the project with UE 4.25.4.

More information on this can be found [here](https://github.com/FS-Driverless/Formula-Student-Driverless-Simulator/blob/master/docs/software-install-instructions.md).

Hope this helps!

KnockerPulsar, 2021-11-06:
I'll try downloading that version and checking if it works. Thank you for the quick reply.

KnockerPulsar, 2021-11-10:
Hello, sorry for the late reply. I can confirm that 4.25.4 works perfectly with this package after following the installation instructions. I even ran a couple of the python examples and they work perfectly. Thank you. 

Just a suggestion: Perhaps you could add previously tested configurations in the readme so people don't have to fiddle with unreal engine versions.

closed 2021-11-10 by KnockerPulsar

wouter-heerwegh, 2021-11-10:
Glad you were able to get it working! I'll try to make some things more clear in the docs, in a couple of days. Feel free to also send in a pull request with some changes you might want.