FS-Driverless/Formula-Student-Driverless-Simulator #280: Outdated documentation about there only being windows binaries available

mateusz-lichota, 2022-01-11:
[The getting started section of the documentation](https://fs-driverless.github.io/Formula-Student-Driverless-Simulator/latest/getting-started/) currently reads

> The simulator is distributed as binaries on every release. At this moment only windows binaries are released. For now, if you are on Ubuntu you will have to run the simulator from the Unreal Engine Editor. During competition, the simulation will run on Windows because it offers a bit better performance and stability.

whereas in fact, there are both windows and linux binaries available for version 2.0.

wouter-heerwegh, 2022-01-12:
I agree, the docs are quite outdated, I'll open a draft pull request to start some changes. Just like updating the getting started guide so that it is clear what people have to do to get the simulator running.

wouter-heerwegh, 2022-01-12:
Just checked, the part you were referring to is not longer in the docs. However, the github pages are behind the latest docs.
I'll try to make sure the latest version of the docs is both in the repo and on the github pages

wouter-heerwegh, 2022-01-12:
Fixed by deploying new version, this version is the same as the docs in this commit: b0dafb6, marked as v2.0.1

closed 2022-01-12 by wouter-heerwegh