FS-Driverless/Formula-Student-Driverless-Simulator #285: UE4 can't compile V2.1.0

MattBrth, 2022-03-03:
When i try to rebuild Blocks and AirSim through UE4 just after runing build.cmd in the AirSim folder I get this issue : 
FSOnline could not be compiled. Try rebuilding from source manually.

This is the log associated : 

How can I fix this ? Thank you !

------------------------------------------------------------------------------------------------------------------

Log file open, 03/03/22 02:37:50
LogWindows: Failed to load 'aqProf.dll' (GetLastError=126)
LogWindows: File 'aqProf.dll' does not exist
LogWindows: Failed to load 'VtuneApi.dll' (GetLastError=126)
LogWindows: File 'VtuneApi.dll' does not exist
LogWindows: Failed to load 'VtuneApi32e.dll' (GetLastError=126)
LogWindows: File 'VtuneApi32e.dll' does not exist
LogConsoleResponse: Display: Failed to find resolution value strings in scalability ini. Falling back to default.
LogConsoleResponse: Display: Failed to find resolution value strings in scalability ini. Falling back to default.
LogInit: Display: Running engine for game: FSOnline
LogPlatformFile: Not using cached read wrapper
LogTaskGraph: Started task graph with 5 named threads and 59 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.201491
LogICUInternationalization: ICU TimeZone Detection - Raw Offset: +1:00, Platform Override: ''
LogInit: Display: Loading text-based GConfig....
LogPluginManager: Mounting plugin MeshPainting
LogPluginManager: Mounting plugin XGEController
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin LiveLink
LogPluginManager: Mounting plugin AISupport
LogPluginManager: Mounting plugin EnvironmentQueryEditor
LogPluginManager: Mounting plugin LightPropagationVolume
LogPluginManager: Mounting plugin AssetManagerEditor
LogPluginManager: Mounting plugin CodeLiteSourceCodeAccess
LogPluginManager: Mounting plugin CLionSourceCodeAccess
LogPluginManager: Mounting plugin GitSourceControl
LogPluginManager: Mounting plugin AnimationSharing
LogPluginManager: Mounting plugin KDevelopSourceCodeAccess
LogPluginManager: Mounting plugin PerforceSourceControl
LogPluginManager: Mounting plugin PixWinPlugin
LogPluginManager: Mounting plugin PluginUtils
LogPluginManager: Mounting plugin PropertyAccessNode
LogPluginManager: Mounting plugin NullSourceCodeAccess
LogPluginManager: Mounting plugin SubversionSourceControl
LogPluginManager: Mounting plugin TextureFormatOodle
LogPluginManager: Mounting plugin PlasticSourceControl
LogPluginManager: Mounting plugin RiderSourceCodeAccess
LogPluginManager: Mounting plugin UObjectPlugin
LogPluginManager: Mounting plugin VisualStudioSourceCodeAccess
LogPluginManager: Mounting plugin VisualStudioCodeSourceCodeAccess
LogPluginManager: Mounting plugin DatasmithContent
LogPluginManager: Mounting plugin XCodeSourceCodeAccess
LogPluginManager: Mounting plugin OodleData
LogPluginManager: Mounting plugin VariantManagerContent
LogPluginManager: Mounting plugin AutomationUtils
LogPluginManager: Mounting plugin OodleNetwork
LogPluginManager: Mounting plugin ChaosClothEditor
LogPluginManager: Mounting plugin BackChannel
LogPluginManager: Mounting plugin AlembicImporter
LogPluginManager: Mounting plugin ChaosCloth
LogPluginManager: Mounting plugin ChaosEditor
LogPluginManager: Mounting plugin ChaosSolverPlugin
LogPluginManager: Mounting plugin CharacterAI
LogPluginManager: Mounting plugin GeometryCache
LogPluginManager: Mounting plugin GeometryCollectionPlugin
LogPluginManager: Mounting plugin GeometryProcessing
LogPluginManager: Mounting plugin MagicLeap
LogPluginManager: Mounting plugin ChaosNiagara
LogPluginManager: Mounting plugin Niagara
LogPluginManager: Mounting plugin MagicLeapLightEstimation
LogPluginManager: Mounting plugin LuminPlatformFeatures
[... 130 lines cut ...]
LogModuleManager: Shutting down and abandoning module PlatformCryptoTypes (42)
LogModuleManager: Shutting down and abandoning module PlatformCrypto (40)
LogModuleManager: Shutting down and abandoning module OodleDataCompressionFormat (38)
LogModuleManager: Shutting down and abandoning module XGEController (36)
LogXGEController: Cannot use XGE Controller as Incredibuild is not installed on this machine.
LogXGEController: Cleaning working directory: C:/Users/berth/AppData/Local/Temp/UnrealXGEWorkingDir/
LogModuleManager: Shutting down and abandoning module AnimationModifiers (34)
LogModuleManager: Shutting down and abandoning module AudioEditor (32)
LogModuleManager: Shutting down and abandoning module PropertyEditor (31)
LogModuleManager: Shutting down and abandoning module TextureCompressor (28)
LogModuleManager: Shutting down and abandoning module RenderCore (26)
LogModuleManager: Shutting down and abandoning module Landscape (24)
LogModuleManager: Shutting down and abandoning module SlateRHIRenderer (22)
LogModuleManager: Shutting down and abandoning module AnimGraphRuntime (20)
LogModuleManager: Shutting down and abandoning module Renderer (18)
LogModuleManager: Shutting down and abandoning module Engine (16)
LogModuleManager: Shutting down and abandoning module CoreUObject (14)
LogModuleManager: Shutting down and abandoning module NetworkFile (12)
LogModuleManager: Shutting down and abandoning module CookedIterativeFile (10)
LogModuleManager: Shutting down and abandoning module StreamingFile (8)
LogModuleManager: Shutting down and abandoning module SandboxFile (6)
LogModuleManager: Shutting down and abandoning module PakFile (4)
LogModuleManager: Shutting down and abandoning module RSA (3)
LogExit: Exiting.
Log file closed, 03/03/22 02:37:57

closed 2022-03-03 by MattBrth

loremacchia, 2022-03-17:
Were you able to solve this? I'm having the same problem

MattBrth, 2022-03-17:
I unfortunately did not. Just clone the Master branch and don't forget to git lfs pull after. It should work.